commaai/openpilot #20596: Hard oversteer on curved road as approaching stop

letsdudiss, 2021-04-06:
**Describe the bug**

When car is about to come to a full stop (normally at around 10-20km/h) and the road is curved/bent, the car steers really hard (causes torque limit warning) into the direction of the curve. i.e if the road is curved to the right, as car slows down to approx. 25km/h, it steers suddenly and hard into the right, causes the car to depart its current lane into the lane on the right

**How to reproduce or log data**

Best reproduced during Stop and Go traffic on windy road

**Expected behavior**

To not steer too hard

**Device/Version information (please complete the following information):**
 - Device: Comma 2
 - Dongle ID: 0ccc2a956a1a1064
 - Route: 0ccc2a956a1a1064|2021-04-06--07-58-46
 - Timestamp: 2m9s (at 18:10:56) and 21m48s (at 18:19:35)
 - Version: 0.8.3 devel (from commaai) with some commits on top (for Stop and Go)
 - Car make/model: Subaru Impreza 2021

**Additional context**

Happens twice, first time is the most obvious. From explorer, the 2 incidents can be easily identified by torque limit warning

pd0wm, 2021-04-08:
Looks like the model was overestimating the turn a little bit. Also at low speed we use quite high steering angles to get you back on the desired path, which quickly hits the torque limit. Hopefully when control aggressiveness is driven by model uncertainty problems like these will get better.

Unfortunately not single easy thing we can fix to make these problems go away.

[image]

letsdudiss, 2021-04-11:
Can i reduce its aggressiveness by default some how?

HaraldSchafer, 2021-04-22:
Yeah low speed control is pretty bad still, and generally too aggressive. It's not a quick to improve that, but it's on the TODO list, thanks for the report!

letsdudiss, 2021-05-28:
Looks like its fixed from this?

https://github.com/commaai/openpilot/commit/dc40631e97739c8e55f271501b6356eda52d84ff

I tried it on my usual route where it used to happen, seems fixed

closed 2021-05-28 by pd0wm