commaai/openpilot #20598: DriverCamera snapshot handling when recordFront is OFF

advcu987, 2021-04-06:
1. Image is extracted even if the `recordFront` setting is OFF (in `snapshot() -> get_snapshots()`). `recordFront` is only assesed at the end of the function and `None` is returned, overwriting the extracted image. 
However, the image should not be extracted at all. code [here](https://github.com/commaai/openpilot/blob/master/selfdrive/camerad/snapshot/snapshot.py#L85)
2. image saving is attempted, even if `fpic` is `None`, see code below, from [snapshot.py](https://github.com/commaai/openpilot/blob/master/selfdrive/camerad/snapshot/snapshot.py#L93) : 

```
pic, fpic = snapshot()
if pic is not None: 
   print(pic.shape) 
   jpeg_write("/tmp/back.jpg", pic) 
   jpeg_write("/tmp/front.jpg", fpic)
```

`jpeg_write` should not be called for`fpic`, in case `fpic` is `None`

pd0wm, 2021-04-06:
1) What does it matter where the image is cleared out? As long as it doesn't leave the `snapshot()` function it's fine, right?
2) The code in `__main__` is only used for testing, it's not called by athena.

advcu987, 2021-04-07:
1. it surely works, just seemed cleaner to avoid the extra steps, hence the enhancement label.
2. got it now, then it's ok, you're right. 

Thanks for the clarification, I'll close the issue, don't want to pointlessly overload.

pd0wm, 2021-04-07:
Actually 1. might be problem since you can also request a message from a socket over athena. If you time it right you might be able to grab something (if it doesn't get blocked upstream due to the large message size) while the snapshot is running. I'll make sure that can't happen!

closed 2021-04-07 by pd0wm