commaai/openpilot #20601: Communication Error Between Processes

speedking456, 2021-04-06:
**Describe the bug**

Communication error appears after approximately 5 minutes into drive. Bug occured today after merging my tuning fork (hehe) with new commits to master.

**How to reproduce or log data**

Go for a drive, engage OP, wait for communication error. I have gotten this error 5 times already, through reboots, and resetting the local repository (and updating submodules).

**Expected behavior**

No communications error.

**Device/Version information**
 - Device: comma two
 - Dongle ID: c75a59efa0ecd502
 - First occurrence: c75a59efa0ecd502|2021-04-06--10-59-12 at 11:06:52
 - Occurs twice in this segment: c75a59efa0ecd502|2021-04-06--11-14-42 at 11:20:53 and 11:23:21
 - Version: https://github.com/speedking456/openpilot - master branch @ e0219c1
 - Car make/model: Kia Seltos 2021

I would have run commaai/master to try it, but my fork apart from tuning has no differences. Also only broke after a pull request from commaai/master, so definitely not my scuffed coding ability.

Created a new branch master-fix removing the merge. It does not have the issue. Drive with that branch: c75a59efa0ecd502|2021-04-06--15-10-18 error did not occur and OP functioned as expected.

Master on my branch has the commits that cause the error.

KDKimura, 2021-04-06:
I too am having this same issue on my comma two stock openpilot 0.8.3 installed in my 2016 Toyota Prius. My DSU is unplugged and community features are enabled. Front camera upload and LDA are enabled. I have uninstalled and reinstalled openpilot and the issue remains. I have to reboot openpilot to resolve the issue, and then restart my car since my DSU is disabled.

adeebshihadeh, 2021-04-06:
@KDKimura can you fill out the bug report template?

KDKimura, 2021-04-07:
Describe the bug

Anywhere from immediately to a few minutes after engaging openpilot, comma two displays error

TAKE CONTROL IMMEDIATELY Communication Issue between Processes

How to reproduce or log data

Uninstall and reintall openpilot if you want because it doesn't solve the issue.
Unplug DSU. I believe the issue also occurred with DSU connected.
Toggle on LDA and front camera upload.
Start car and engage openpilot.
Sometimes it occurs immediately, other times it occurs within 4 minutes.

Expected behavior

Less energy expenditure in the arms and Wall-E like obliviousness....openpilot should function without this error.

Device/Version information (please complete the following information):

Device: comma two
Dongle ID: 55c431eb30c49edb
Route: 55c431eb30c49edb|2021-04-05--13-04-24
Timestamp: 0-5 minutes into the drive
Version: 0.8.3
Car make/model: Toyota Prius 2016
Additional context

🤬

pd0wm, 2021-04-07:
None of the mentioned routes are completely uploaded. If you upload them I'll look into it. I assume it's the average frequency check in submaster, but would be interesting to see which message isn't running at the right frequency.

pd0wm, 2021-04-07:
Probably fixed by: https://github.com/commaai/openpilot/commit/0dca8cb7dd2e515beb955ad8cb146f6a5b5f7941. There were still some leftovers from the gpsMalfunction alert.

speedking456, 2021-04-07:
> Probably fixed by: [0dca8cb](https://github.com/commaai/openpilot/commit/0dca8cb7dd2e515beb955ad8cb146f6a5b5f7941). There were still some leftovers from the gpsMalfunction alert.

Will merge with my branch and test

KDKimura, 2021-04-07:
> None of the mentioned routes are completely uploaded. If you upload them I'll look into it. I assume it's the average frequency check in submaster, but would be interesting to see which message isn't running at the right frequency.

How do I upload my routes? Does Comma Connect not do that automatically?

speedking456, 2021-04-07:
> How do I upload my routes? Does Comma Connect not do that automatically?

I believe they were talking about my routes, something is up with my comma prime sim, as the carrier I'm using only has 128 kbps for it's unlimited data. I'm in Canada, and using lucky mobile prepaid for it.

I'll make sure they get uploaded by the end of today, somehow.

speedking456, 2021-04-07:
> Probably fixed by: [0dca8cb](https://github.com/commaai/openpilot/commit/0dca8cb7dd2e515beb955ad8cb146f6a5b5f7941). There were still some leftovers from the gpsMalfunction alert.

Did not fix the issue, still occurred during my latest drive: c75a59efa0ecd502|2021-04-07--11-20-16

pd0wm, 2021-04-07:
You need to connect to WiFi for the full logs to upload.

speedking456, 2021-04-07:
> You need to connect to WiFi for the full logs to upload.

I was, all night long. None of my logs have uploaded fully with the latest commits. However, they do appear to upload fine before the commits, so idk.

pd0wm, 2021-04-08:
Logs have uploaded. Looks like paramsd (a non realtime process) didn't get scheduled for 0.87 seconds while there was activity from the modem. This is one of our longer standing issues we're still looking into. For most people it's a very rare issue, but some people seem to reliably hit it based on something like cell tower location. If the issue persists you can try removing the SIM card from your device.

KDKimura, 2021-04-08:
> Logs have uploaded. Looks like paramsd (a non realtime process) didn't get scheduled for 0.87 seconds while there was activity from the modem. This triggers an internal health check forcing a soft disengagement.
> 
> 
> 
> This is one of our longer standing issues we're still looking into. For most people it's a very rare issue, but some people seem to reliably hit it based on something like cell tower location. If the issue persists you can try removing the SIM card from your device.

Whose log are you speaking about?

pd0wm, 2021-04-08:
@speedking456's logs.

If you have a specific drive for me to look into I can do that as well.

pd0wm, 2021-04-12:
Closing this issue. Feel free to open a new one if you still have problems with a new route.

closed 2021-04-12 by pd0wm