commaai/openpilot #20606: Toyota: Cruise Fault Malfunction on 21 Camry

Nyck310, 2021-04-07:
**Describe the bug**

2021 Camry Nightshade. Every time C2 boots up to the camera screen, I get a "Cruise Fault" Error on my OP HUD and am unable to use cruise control or OP.

**How to reproduce or log data**

**Expected behavior**

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID:  2702185606b65689
 - Route: 2702185606b65689|2021-04-06--19-15-57
 - Timestamp: Immediate
 - Version: Master ( hash a99a55ce2b), also 0.8.3
 - Car make/model: Toyota Camry Nightshade 2021

**Additional context**

adeebshihadeh, 2021-04-07:
Can you record a route with the stock system and post it here? Make sure the toggle in the off position in Settings -> Toggles -> Enable openpilot. The route only needs to be a minute or two.

Nyck310, 2021-04-07:
I'll edit original as well, here it is:
2702185606b65689|2021-04-07--13-59-10

adeebshihadeh, 2021-04-07:
The logs aren't fully uploaded from that route. Can you leave your device connected to wifi for a bit so they can upload?

Nyck310, 2021-04-07:
Understood. connecting now. I keep it inside since that AZ sun is kinda brutal on it.
should be uploaded in 5.

adeebshihadeh, 2021-04-11:
The error you're getting now is unrelated to #19731, but it's the same as #20207.

Nyck310, 2021-04-11:
Understood, is that fix in progress as well or has it hit a barrier?

adeebshihadeh, 2021-04-12:
We'll look into this before the next release.

anotabien, 2021-04-14:
One thing to add that it happens to my Brazil made Corolla 2021 non-hybrid.
With original parameter, most of the time will get ¨CRUISE FAULT: Restart your car¨ (80/90% of the time) and problem will not be solve by restarting the car. When it starts with out the ¨CRUISE FAULT: Restart your car¨, OP will work fine with Stop&Go and can be engage in any speed.
With ¨LOW_SPEED_LOCKOUT = False¨, ¨CRUISE FAULT: Restart your car¨ won´t came out, but looses Stop&Go feature, only can be engage over 22km/h.

adeebshihadeh, 2021-04-20:
Moving this issue over to #20703 since it has more information. Also added @Nyck310's ifno to #20703.

closed 2021-04-20 by adeebshihadeh