commaai/openpilot #20617: When starting Corolla 2017, engine may stutter + comma2 reboots immediately

billma12, 2021-04-08:
**Describe the bug**

After updating to 0.8.3, my Corolla 2017 SE engine starts to stutter when starting the car + comma2 reboots. The comma2 will eventually turn back on while I'm driving and openpilot is functional. 

**How to reproduce or log data**

Not always reproducible, but seems like this issues occurs after being away from car for a couple hours, e.g when I go to my car in the morning this issue will happen.

**Expected behavior**

When car starts, engine should not stutter and comma2 should not reboot

**Device/Version information (please complete the following information):**
 - Device: comma2
 - Dongle ID: da8078584fafb6c1
 - Route: da8078584fafb6c1\|2021-04-07--19-14-54
 - Timestamp: At the very beginning of the drive (around 7:10pm which is before the timestamp of the video start)
 - Version: stock 0.8.3
 - Car make/model: Toyota Corolla 2017

**Additional context**

- Toyota Pedal is attached
- first observed this on 0.8.3
-  After my drive was over I immediately tried testing this again (turning car off and on) but this time issue did not occur: da8078584fafb6c1|2021-04-07--19-22-44

pd0wm, 2021-04-08:
Did you try without the pedal? And without the C2 at all? This sounds more like a car issue to me.

DarkSlaayer, 2021-04-08:
> Did you try without the pedal? And without the C2 at all? This sounds more like a car issue to me.

Yeah sounds like a weak battery. @billma12 might go by an auto parts store and see if they can test your battery.

billma12, 2021-04-08:
Thanks that makes sense, I unplugged the C2 + pedal and still experience this.

My question is what would cause the C2 to reboot when the car eventually starts up though?

DarkSlaayer, 2021-04-08:
> Thanks that makes sense, I unplugged the C2 + pedal and still experience this.
> 
> My question is what would cause the C2 to reboot when the car eventually starts up though?

Voltage drop. C2 does not have a battery.

billma12, 2021-04-09:
I'll go get my car checked and close this if that resolves the issue.

pd0wm, 2021-04-09:
Since it happens without the C2 as well I'm going to close the issue. Feel free to open a new issue if your C2 is still misbehaving after your mechanical problems are solved.

closed 2021-04-09 by pd0wm