commaai/openpilot #20648: "Car Unrecognized"

CPontes, 2021-04-11:
I just installed comma Two in my corolla 2021 and it appears Car Unrecognized 

The camera works well, it simulates the route, but doesn't recognize the car. 

 - Device: comma two
 - Dongle ID: cb7a33a4e4f4116b 
 - Route: every route 
 - Timestamp: all the time 
 - Version: open pilot v0.8.3 
 - Car make/model: Toyota Corolla hybrid 2021 [e.g. Toyota Prius 2016]

**Additional context**

I already reinstall open pilot, but it didn't work, 

Thanks in advance for the help.

vanillagorillaa, 2021-04-11:
https://github.com/commaai/openpilot/pull/20650 should fix your issue

adeebshihadeh, 2021-04-11:
Merged @vanillagorillaa's PR into master. Here's a guide to get onto a branch where your car is supported - https://github.com/commaai/openpilot/wiki/Update-Modify-openpilot#how-to-switch-to-master-ci-non-developers. If you need help, join our [discord[(https://discord.comma.ai).

closed 2021-04-11 by adeebshihadeh

CPontes, 2021-04-11:
Hello, I did the following steps but still didn't work. Now it doesn't open the camera any more. 
It's running openpilot v0.8.4-337ac2b4