commaai/openpilot #20661: Cruise Fault - Toyota Corolla Hybrid 2021 v0.8.4-337ac2b4

CPontes, 2021-04-12:
Hello Community, :)

I SSH'ed into the system
updated it to current master
it rebooted.
confirmed that It's running openpilot v0.8.4-337ac2b4

When connect comma in the car, it runs the camera but appears the message "Cruise Fault: Restart car to engage"

I restarted the car, unplugged and plugged comma, but still didn't work. 

Device: comma two
Dongle ID: cb7a33a4e4f4116b
Route: every route
Timestamp: all the time
Version: open pilot  v0.8.4-337ac2b4
Car make/model: Toyota Corolla hybrid 2021

Thank you so much for the attention in advance.

[image]

adeebshihadeh, 2021-04-12:
Did you always have this fault or did openpilot work at some point on your car?

CPontes, 2021-04-13:
> Did you always have this fault or did openpilot work at some point on your car?

Hi, @adeebshihadeh, thanks for the message. It's the. first time I'm installing comma.ao, first it wasn't recognizing the car. After I SSH the system and updated to open pilot v0.8.4-337ac2b4, now it recognizes the car, opens the camera, but It's happening this fault, and openpilot didn't work on my car yet. 
Thank you so much for the help

adeebshihadeh, 2021-04-13:
Closing this issue since it's a duplicate of #20606. We're going to look into this before the next release.

closed 2021-04-13 by adeebshihadeh

anotabien, 2021-04-14:
One thing to add that it happens to my Brazil made Corolla 2021 non-hybrid.
With original parameter, most of the time will get ¨CRUISE FAULT: Restart your car¨ (80/90% of the time) and problem will not be solve by restarting the car. When it starts with out the ¨CRUISE FAULT: Restart your car¨, OP will work fine with Stop&Go and can be engage in any speed.
With ¨LOW_SPEED_LOCKOUT = False¨, ¨CRUISE FAULT: Restart your car¨ won´t came out, but looses Stop&Go feature, only can be engage over 22km/h.