commaai/openpilot #20667: Overly loud warnings for steering unavailable

afoerster, 2021-04-13:
**Describe the bug**

On master branch I'm getting very loud 'steering unavailable' warnings when going around corners.  This happens quite often, the route below has a few instances in two miles. The noise is jarring.

**How to reproduce or log data**

Happens driving around corners, or entering a corner.

**Expected behavior**

A less noisy/intrusive warning when steering is unavailable driving through corners. 

**Device/Version information (please complete the following information):**
 - Device: Comma2
 - Dongle ID: 6a8fa21108f02593
 - Route: 6a8fa21108f02593%7C2021-04-11--08-41-17--0
 - Timestamp: Several times within the first few minutes of the route
 - Version: master 337ac2b4c50c0c30ca83f036fe08ce89bf6caae0
 - Car make/model: 2019 Acura RDX Tech

**Additional context**

Initially discussed with @adeebshihadeh in https://github.com/commaai/openpilot/pull/20656, where a cause was pointed out: https://github.com/commaai/openpilot/pull/20567

VirtuallyChris, 2021-04-13:
Can attest to this. The new changes are almost unusable on Honda / Acura vehicles since steer unavailable happens every time the wheel is slightly jerked while engaged.

adeebshihadeh, 2021-04-21:
We still want to alert for loss of steering, but made the alert less harsh when overriding in #20718.

closed 2021-04-21 by adeebshihadeh