commaai/openpilot #20675: Take Control immediately, communication issue between processes

defyent, 2021-04-13:
Twice today, at what seemed to be random times (first was within 10 minutes of the drive, and probably 5 minutes of initiating OP, second was about 30 minutes into a different drive, and within 10 minutes of initiating OP) I got the alarm "Take Control immediately, communication issue between processes". I took over immediately and disengaged OP, but was able to immediately reengage OP both times. 

**How to reproduce or log data**

Seemingly random times, nothing eventful happened to trigger the alarm that I noticed.

**Expected behavior**

Not to alarm

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: cc5727706f6dd3c3
 - Route: [e.g. cc5727706f6dd3c3|2021-04-12--13-17-14
 - Timestamp: around 09:07:00 timestamp
 - Version: 0.8.3
 - Car make/model: Toyota Corolla Hatchback Hybrid 2018

**Additional context**

I've found the other bug reports that are closed from a few days ago, one mentioned commit 0dca8cb, I looked at the change and the code on my device and it seems my device already has this commit applied, "ubloxRaw" does not exist anywhere in the controlsd.py file.

speedking456, 2021-04-14:
I recently ran into this issue as well, as did @KDKimura. Apparently this is a long standing issue—pd0wm mentions it in my issue #20601—however only recently has this actually become an issue for me. Just trying to find similarity, but are you using a sim card in your device?

defyent, 2021-04-14:
Hrm okay, nope no sim card, wifi only.

speedking456, 2021-04-14:
> Hrm okay, nope no sim card, wifi only.

Alright well, pd0wm's solution of removing the sim card won't work for you. Are you able to go into developer of your comma two and check what commit you are running?

defyent, 2021-04-14:
9a0006fa0a

pd0wm, 2021-04-14:
Looks like `liveParameters` and `liveLocationKalman` lagged for about a second triggering the alert. My current best guess it this is caused by the garbage collector. This is turned off for for most processes like controlsd, but looks like it's still enabled for locationd and paramsd. 

We're currently rewriting locationd and paramsd in C++, which should of course solve this problem. I'll see if I can disable the garbage collector in the meantime.

[image]

closed 2021-04-15 by adeebshihadeh

pd0wm, 2021-04-15:
Think this specific case should be fixed now. Feel free to open a new issue when you run into the problem again.