commaai/openpilot #20682: OP doesn't recognized the side of a car

edtreo, 2021-04-14:
**Describe the bug**

This morning while OP is slowing down appropriately toward a stopped vehicle, a car coming from an alley decided to cross the street from my right to make a left turn. OP complete ignored the side of the car and decided to speed up as if there is nothing in front of me.
**How to reproduce or log data**

see above
**Expected behavior**
To recognize a vehicle whether it's the back or side of it

**Device/Version information (please complete the following information):**
 - Device: [comma two]
 - Dongle ID: [ff1be2c236f83345]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [4/14, around 9:30]
 - Version: [0.8.3]
 - Car make/model: [e.g. Toyota Rav4 2021]

**Additional context**

pd0wm, 2021-04-15:
Can you fill out the template? We need to know the route to look into it:
 - Device: [comma two]
 - Dongle ID: [ff1be2c236f83345]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [4/14, around 9:30]
 - Version: [0.8.3]
 - Car make/model: [e.g. Toyota Rav4 2021]

edtreo, 2021-04-15:
ff1be2c236f83345	ff1be2c236f83345|2021-04-14--09-26-36	explorer	preserve	3.97579	True	True	12	12	12	12	12	12	12	TOYOTA RAV4 2019	https://github.com/commaai/openpilot.git	release2	False

Sent from my iPhone

> On Apr 15, 2021, at 4:41 AM, Willem Melching ***@***.***> wrote:
> 
> my.comma.ai/useradmin

pd0wm, 2021-04-15:
Crossing cars are not guaranteed to be classified as a lead car in your lane by the model. These situations will have to be fixed by end-to-end longitudinal.

closed 2021-04-15 by pd0wm