commaai/openpilot #20689: OP pulls towards curb due to improper curb detection

abdulrahman004, 2021-04-15:
Driving in curb lane which doesnt have a solid lane line on curb edge. left lane line is also pretty bleak.  I believe I was driving along a section where there was no raised curb, which OP probably thought the lane widened and pulled towards right. Had to counter steering

Expected behavior
Car shouldn't have pulled towards curb

Device: FreEon
Dongle ID: 59f457ce47259dd2

Route: 59f457ce47259dd2|2021-04-15--03-04-41--4
Timestamp: 20:09:01
Version: 0.8.3 (release 2)
Car make/model: Toyota Sienna 2018
Additional context
gif of the report
https://imgur.com/a/qPVDeUm

HaraldSchafer, 2021-04-22:
This is a pretty weird one, I will have to investigate

Slevner, 2021-04-28:
Russia - Moscow, similar behavior. Often sees the middle of the curb as a strip. of 0.8.3 the situation is the same.

calledit, 2021-05-04:
I have that problem too. If i remember correctly it works better when you turn on laneless.

davetapley, 2021-08-04:
All the time [on this street](https://www.google.com/maps/@33.4648372,-112.0652269,3a,75y,179.45h,78.87t/data=!3m6!1e1!3m4!1se3loPQLNJM6mqeSZ5HUL9g!2e0!7i16384!8i8192) in Phoenix, AZ, USA ⬇️. I have to pull the wheel to the left else it just drifts closer and closer 😱 

[image]

I can get some routes/segments if useful?

HaraldSchafer, 2021-08-09:
We are only really actively working on improving laneless performance at this point.

closed 2021-08-11 by HaraldSchafer