commaai/openpilot #20703: "CRUISE FAULT" error on TSS2 Toyotas non-US market

tomascw, 2021-04-19:
**Describe the bug**

"CRUISE FAULT" error 

**How to reproduce or log data**

Starting the car on 90% of the time

**Expected behavior**

The expected behaviour is to handle the "CRUISE FAULT" error, considering this cars have a specific fingerprint, differing from other TSS2 Toyota cars that doesn't have this kind of error.

**Device/Version information (please complete the following information):**

 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

@alfatag
 - Device: Comma two
 - Dongle ID: 4aef35a12dbe0a01
 - Route: Not applicable
 - Timestamp: Not applicable
 - Version: Shane's based on 0.8.2, Openpilot master-ci and 0.8.3
 - Car make/model: Corolla Hybrid 2021 (Brazil made - Brazilian market)

@Anotabien
- Device: Comma 2
- Dongle ID: 1e9d51d3eb083940
- Route: Not applicable
- Timestamp: Not applicable
- Version: OP 8.1 master-ci/OP 8.2 release/OP 8.3 release/Shane Fork 0.6.6(8.2)
- Car make/model: Toyota Corolla non-hybrid 2021 (Brazil made - Argentina market)

@Ale Sato Brazil SP
- Device: EON Gold
- My Dongle: 7d5fbee9d4476ca3
- Route: Not applicable
- Timestamp: Not applicable
- Car make/model: Toyota / Corolla Hybrid 2020 (Brazil made - Latinamerican market)

@tomascw
- Device: Comma 2
- Dongle ID: 1a5384cb44ae8c65
- Route: Not applicable
- Timestamp: Not applicable
- Version: 0.8.2 / 0.8.3
- Car make/model: Toyota Rav4 HV Limited 2019 (Japan made - Argentina market)

@tape
 - Device: comma two
 - Dongle ID: 5dc0b649528bd49c
 - Route: 5dc0b649528bd49c|2021-04-17--07-04-17
 - Timestamp: Not applicable
 - Version: 0.8.3
 - Car make/model: Toyota Corolla Cross Hybrid (Thailand made and Asian market eg Thailand, Vietnam and Taiwan)

@fabiomla
- Device: comma two
- My Dongle: 66334e164c2e9976
- Route: Not applicable
- Timestamp: Not applicable
- Car make/model: Toyota / Corolla Hybrid 2020 (Brazil made - Latinamerican market)

**Additional context**

We understand all cars are TSS2 with no LTA and Stop and go in the Toyota Stock features.

Of the 10% of the times it works, openpilot works perfectly, including "stop and go".

As a workaround, the "CRUISE FAULT" problem is by changing the "LOW_SPEED_LOCKOUT = False", where openpilot works 100% of the time when starting the car, but with no "Stop and Go" functionality.

adeebshihadeh, 2021-04-20:
This is a duplicate of #20606, but I'll leave this one open since it has more info. From #20606:

Nyck310, 2021-05-22:
For clarification, my car (21 Camry) is a US model.

ntegan, 2021-05-27:
This is affecting my US 2017 Corolla LE.
I had it working fine (DSU-less) for a day or so and now I get cruise fault every time.
Can comment out low speed lockout event in selfdrive/car/toyota/interface.py but then I don't get full stopping to 0mph, it just disengages at 25-28mph

When I had it working, OP would break to 0mph with a lead car (but not start from 0mph as i don't have the pedal)

ErichMoraga, 2021-05-27:
> This is affecting my US 2017 Corolla LE.
> I had it working fine (DSU-less) for a day or so and now I get cruise fault every time.
> Can comment out low speed lockout event in [...]

Your DSU-based TSS-P issue is unrelated to this TSS2 issue.

zorrobyte, 2021-05-31:
Got another one here:
https://discordapp.com/channels/469524606043160576/524327905937850394/848738053446697010

@adeebshihadeh it may be worth removing, or at least adding a note to the readme that 2021 Camry support is hit and miss, even on US model spec cars.

daehahn, 2021-06-07:
It is always happened with my US version of the 2018 Highlander Hybrid. Solution for me is 'turn off engine and restart'. My car is clearly not a TSS2. I could not remember exact number, but it was started from some versions ago. With Comma-Power, it didn't happen on old versions. When I figure out more. I will update my comments here.

ErichMoraga, 2021-06-07:
Yeah, but was your TSS-P Highlander's DSU disconnected?  Really not a good idea to clog up this TSS2 issue.

daehahn, 2021-06-07:
it may not a TSS2 issues. That is the reason why I post my comments here.

daehahn, 2021-06-07:
Found the reason at least for my car. I will reach to other folks for the verification. I will be back to this thread when finished.

zorrobyte, 2021-06-09:
I had something like this, but for an unrelated issue (using gray panda+giraffe on TSS2 Rav4)

One can disable OP long and revert to factory by forwarding the ACC messages & disabling OP long: https://github.com/zorrobyte/openpilot/tree/TSS2_stockCruise

ShawnOnGitHub, 2021-07-02:
Device: comma two
My Dongle: dd6ca38aa430a72e
Route: Not applicable
Timestamp: Not applicable
Car make/model: Toyota / Camry LE 2021 **US Model**

"Cruise Fault" 100% time
OpenPilot can't engage
Restart the car doesn't solve the issue

Nyck310, 2021-07-02:
> Device: comma two
> My Dongle: dd6ca38aa430a72e
> Route: Not applicable
> Timestamp: Not applicable
> Car make/model: Toyota / Camry LE 2021 **US Model**
> 
> TSS 2.5+
> "Cruise Fault" 100% time
> OpenPilot can't engage
> Restart the car doesn't solve the issue

The last time the restarting trick worked for 21 camry was on 0.8.3.
That still does involve turning key till you  see the cruise fault error, and then back, and repeatig every 10 seconds until you don't  get the error.

For long trips, its work it ( normally takes 2 or 3 min till no error and you can start car), otherwise, wait until 21 camry is supported again or return for refund.

gazeek, 2021-07-02:
Exactly same problem as @ShawnOnGitHub 
My Dongle: 65e48933dc1b498a
Route: Not applicable
Timestamp: Not applicable
Car make/model: Toyota / Camry LE 2021 **US Model**

TSS2.5+
"Cruise Fault" 100% time
OpenPilot can't engage
Restart the car doesn't solve the issue.

The following attempts were unsuccessful:

- To restart the car using some magic as suggested on:
   https://discord.com/channels/469524606043160576/524327905937850394/816110647048667137
- Try release branches 0.8.4, 0.8.5
- I also tried a bunch of branches, for which I always ran: git submodule init && git submodule update && scons -j4 && reboot
  - Stock Cruise fork from zorrobyte:
     Cloned from https://github.com/zorrobyte/openpilot.git and using branch TSS2_stockCruise.
  - Someone who claimed that this is fixed in a github issue https://github.com/commaai/openpilot/pull/21220:
     Cloned from https://github.com/daehahn/openpilot.git using branch fix-can_recv
  - I tried the original Camry 2021 merge to master commit:
    https://github.com/commaai/openpilot/tree/ff0be15bbc382d13c565132342cb50f2e3a342da

I was going to try relogger.py on the logs that I collected when the car was starting up, but I didn't have time yet to setup my ubuntu environment to make it all work together and on my mac I am still trying to get all the python dependencies to work.
I can submit all the logs if that would help (it's about 1.2GB from all the tries).

ShawnOnGitHub, 2021-07-02:
> Exactly same problem as @ShawnOnGitHub
> My Dongle: 65e48933dc1b498a
> Route: Not applicable
> Timestamp: Not applicable
> Car make/model: Toyota / Camry LE 2021 **US Model**
> 
> TSS2.5+
> "Cruise Fault" 100% time
> OpenPilot can't engage
> Restart the car doesn't solve the issue.
> 
> The following attempts were unsuccessful:
> 
> * To restart the car using some magic as suggested on:
>   https://discord.com/channels/469524606043160576/524327905937850394/816110647048667137
> * Try release branches 0.8.4, 0.8.5
> * I also tried a bunch of branches, for which I always ran: git submodule init && git submodule update && scons -j4 && reboot
>   
>   * Stock Cruise fork from zorrobyte:
>     Cloned from https://github.com/zorrobyte/openpilot.git and using branch TSS2_stockCruise.
>   * Someone who claimed that this is fixed in a github issue [Toyota: 'Cruise Fault' fix when ignition. Not a TSS2 related fix. #21220](https://github.com/commaai/openpilot/pull/21220):
>     Cloned from https://github.com/daehahn/openpilot.git using branch fix-can_recv
>   * I tried the original Camry 2021 merge to master commit:
>     https://github.com/commaai/openpilot/tree/ff0be15bbc382d13c565132342cb50f2e3a342da
> 
> I was going to try relogger.py on the logs that I collected when the car was starting up, but I didn't have time yet to setup my ubuntu environment to make it all work together and on my mac I am still trying to get all the python dependencies to work.
> I can submit all the logs if that would help (it's about 1.2GB from all the tries).

You sir saved people a lot of time. 
I understand car support is not comma ai top security. But this issue pretty much render Camry 21 incompatible. I wonder if Shane or someone can provide a projected timeline on fixing this bug?
I will post my progress here.

sshane, 2021-07-02:
@ShawnOnGitHub @gazeek can you turn on upload raw logs in settings and let your devices sit on WiFi until they are uploaded? Also please post a route name with the issues

sshane, 2021-07-02:
Also it would be helpful if someone with a 2021 Camry uploaded a route with `Enable openpilot` in settings toggled off. Make sure to engage for a bit while using the stock system as well.

gazeek, 2021-07-04:
@sshane I have gotten the data you requested:
* log with `Enable openpilot` set to on:
  * `65e48933dc1b498a|2021-07-04--13-20-18--0`
* log with `Enable openpilot` set to off:
  * `65e48933dc1b498a|2021-07-04--13-28-22--0` 
  * `65e48933dc1b498a|2021-07-04--13-37-22--0`

Please note that `Cruise Fault` shows up immediately after turning the ignition to on. 
I can perform more experiments as you request, but it might take me till the weekend to provide the logs.

ShawnOnGitHub, 2021-07-05:
@sshane 
My C2 has only a few routes recorded/uploaded. Same as  @gazeek, all of them demonstrate the bug since car started. OP can't engage. 

dd6ca38aa430a72e

I can help experiment it as well. Please let me know.

bgtvoy205, 2021-07-05:
Me too, I also am having  the same problem!! 
[image]

CreekRun, 2021-07-05:
Same tell me how to post a bounty? And I will do 500$ for the person who fixes this issue in the next 24hrs!?

JoshuaHarrisJBh, 2021-07-05:
Oh this issue has been going on for a while, guess I overlooked this before. Really don't know anything about coding, though I'm tech savvy enough to figure most problems out. This I can not. Help! Please! I assumed by having a car that was on the comma ai supported list that it would be easy. 
There's 4 days before the 30 day return period is up. 
I don't want to return it,seeing as it was the biggest reason behind the model of car I went with!!
Not to mention I bought the red edition last night! 
If someone will tell me how long it will be, or how I can help that would be awesome. 
People that support comma are all fans, especially when they offer money in exchange for a solution to the problem.

sshane, 2021-07-07:
Thanks for the routes gazeek. @Nyck310 did you say you can reproduce stop and go functionality if you stop and restart the car a few times? A route where openpilot is able to accelerate from a stop and has no faults would be helpful!

Nyck310, 2021-07-07:
2702185606b65689|2021-07-05--12-27-27

Long ride fully functional OP so I'm  sure it includes stop n go. 0.8.3 is the latest version I can use the Restart Workaround for OP, anything later I've  been SOL.

sshane, 2021-07-07:
@Nyck310 Make sure to have upload raw logs enabled and let the device sit for the logs to upload.

Nyck310, 2021-07-07:
No raw logs on 0.8.3. At least, no option on the GUI.
This drive is from a few days ago so I can see it all uploaded myself, but I'll make it public just in case.  If anything else, please advise.

sshane, 2021-07-08:
You want to make sure that the `None`s in the third column are replaced with links to download rlogs in useradmin: https://my.comma.ai/useradmin/?onebox=2702185606b65689%7C2021-07-05--12-27-27. If your device is online you can also see a button to request upload at the bottom of the list.

Nyck310, 2021-07-08:
I'll  check when I go back to the garage but I'm  not sure I remember a request upload button on 0.8.3.
If I don't  respond back tonight, I'll  do more on my lunch break tomorrow.

Cheers

Nyck310, 2021-07-08:
results: 2702185606b65689|2021-06-18--12-18-22
Older, but all four filled.
Hope this helps!

sshane, 2021-07-08:
That one looks good, thanks for the route! One more question, if you ignore the `lowSpeedLockout` and a fault occurs, can the car brake to zero and just not accelerate, or does it fully disengage under a certain speed?

Nyck310, 2021-07-08:
I've  never modified to use that sng workaround so I'm not sure tbh, I'm  pretty stacked until monday, but if no one can post a route for that by then, I can try and see what exactly happens.

tomascw, 2021-07-08:
I can say for me (and the others Non-US toyotas) that the car disengage at
the same speed as the ACC system without the OP. In my case 27km/h.

sshane, 2021-07-08:
Would someone want to test this branch? https://github.com/commaai/openpilot/pull/21512 It should not give a cruise fault error, but stop and go (full range DRCC) will not work as it's not a feature from the factory. I'm interested to know if it brakes to zero though. Let me know your results!

sshane, 2021-07-09:
@Nyck310 @ShawnOnGitHub @gazeek can one of you test out https://github.com/commaai/openpilot/tree/low-speed-lockout? It should be ready now!

Nyck310, 2021-07-09:
I'll  be able to test in about 6 hours on 21 camry.

Nyck310, 2021-07-10:
2702185606b65689|2021-07-09--16-34-03   -- low-speed-lockout

So the cruise fault error is officially gone, which is amazing, but also SnG is gone as well...  when speeds drop to 15mph it disengages. Progress, but I'll probably revert back to 0.8.3 for my trip tomorrow so I can have SnG during traffic.
I'll try to be available for tests over the weekend, but definitely a little tonight and Monday.
Thanks for your effort!

sshane, 2021-07-10:
@Nyck310 Thanks for testing! Your logs aren't fully uploaded for what ever reason (try to leave the device on for a few extra minutes), but another user's car disengages at 12 mph, so I imagine yours might as well. Do you have a Discord? I have another branch with an experiment to try and enable sng in recent versions of openpilot (won't get merged any time soon, but interested to see if it works).

adeebshihadeh, 2021-07-10:
This should be fixed by 16e4f432bf592ad74dcd6085e55c01570d642757. The fix will be included in the next release.

closed 2021-07-10 by adeebshihadeh

Nyck310, 2021-07-10:
Wait... Sorry, why is this closed, isn't  it the same result as changing only the low speed lockout to false? SnG is still screwed and nothing is different...
If this fixed the other non US cars awesome, though should I reopen a ticket for 21 camrys?

sshane, 2021-07-10:
@Nyck310 Only certain trims of the Camry has a permanent low speed lockout, and we definitely don't want to ignore the lockouts on all cars, where it being non-2 is a valid lockout and we shouldn't engage. That PR correctly detects if the car has a permanent low speed lockout (no full range ACC from the factory) and ignores it since it's not a valid reason to lock you out from openpilot.

The patch I'm thinking might enable stop and go won't be merged to upstream (since these cars don't have an electronic parking brake, which is usually required by the manufacturer for stop and go), but you can feel free to incorporate it on any fork you run. I'll message you on Discord more info

gazeek, 2021-07-12:
@sshane thanks for fixing this issue. I have also verified (albeit a bit late...) that it works as expected in log `65e48933dc1b498a|2021-07-11--21-20-00--0`

Getting stop and go working for me would be something I am interested in pursuing (as does @Nyck310, it would seem ), so I might spend time trying to get it to work on it on the weekends since comma does not want to spend time on it.

My only request is that the README.md on master branch be updated for Camry 2021 (and maybe others), currently for Camry 2021 it says:
| Make      | Model (US Market Reference)   | Supported Package | ACC              | No ACC accel below | No ALC below      |
| ----------| ------------------------------| ------------------| -----------------| -------------------| ------------------|
Toyota | Camry 2021 | All | openpilot | 0mph | 0mph

From what you said, there should be some tag <sup>5</sup> that reads something like:
 <sup>5</sup> 15mph for models which don't have Full-Speed Range Dynamic Radar Cruise Control, see [TSS Applicability Chart](https://engage.toyota.com/static/images/toyota_safety_sense/TSS_Applicability_Chart.pdf)

tomascw, 2021-07-12:
> From what you said, there should be some tag 5 that reads something like:
> 5 15mph for models which don't have Full-Speed Range Dynamic Radar Cruise

Yes. The same for thE TSS2 Toyotas without full ACC like my RAV4 2019
Hybrid.