commaai/openpilot #20704: Large Shadow phantom braking

8a9phrywet978ui, 2021-04-19:
**Describe the bug**

Tall vehicles when casting a shadow into the adjacent lane cause openpilot to mis-identify the shadows as vehicles, and abruptly brake even with no actual vehicle in front of you in your lane

**How to reproduce or log data**

Very Large Vehicle or Semi in Right Lane, openpilot vehicle in Left Lane,  sun casting shadow of the Semi onto majority of left lane 

**Expected behavior**

Not doing this

**Device/Version information (please complete the following information):**
 - Device: Comma 2
 - Dongle ID: 9d8c378664ae56f2
 - Route: No Longer Have, multiple instances on April 12th throughout my 14 hour drive.
 - Timestamp: N/A, after Noon
 - Version: Release2-staging
 - Car make/model: 2019 Toyota Corolla Hatchback

**Additional context**

Other users reporting this issue on the discord, hopefully one of them can provide a better example

pd0wm, 2021-04-20:
Without a route this is pretty hard to look into for us. Let us know when you have an example.

ShaneSmiskol, 2021-04-20:
I've noticed similar behavior on the latest KL model. It's super eager at picking out leads, when coming up on stopped traffic in multiple lanes it will quickly switch between the adjacent lanes until I get close enough. This also happens when there's no traffic in my lane, so it will brake for a moment until it realizes there's no lead.

adeebshihadeh, 2021-06-17:
Closing this due to inactivity. Let us know if you have a route for us to look into.

closed 2021-06-17 by adeebshihadeh

Achilles308, 2021-09-10:
> Closing this due to inactivity. Let us know if you have a route for us to look into.

I reported a shadow bridge braking example on discord some 4 weeks or so ago. I tried to find the bridge that it happened at, but with so many to pick from and no readily easily available brake logs on cabana, I was not able to find it. It broke from 70 to 65... so minor... but still a braking event for no reason.

This even was a braking event after a street light and into the shadow behind it (best I can tell, that is what I think it was braking for at least). The set speed was 40 and sometime around the streetlight or the shadow, OP broke to 35.

comma 3
master-ci 0.8.10
dfb993cc43ba69dc|2021-09-09--21-50-02--18 and 19
22:08::57 Local time
2020 Toyota Corolla Hatchback

Please reopen this, as it does happen from time to time and I'd like to log them better. It would be helpful to know if this is a VOACC or a radar issue?