commaai/openpilot #20705: Use exposure for screen brightness

VirtuallyChris, 2021-04-19:
Sometimes, a bad calibrated light sensor can get in the field. It would be much better if we use the camera exposure while onroad instead of the light sensor. Likely would fix things like https://github.com/commaai/openpilot/issues/20419 since the light sensor values vary from device to device.

adeebshihadeh, 2021-04-20:
This wouldn't fix #20419, from @gregjhogan

> forced brightness to always be the max and it didn't help, the overlays/sidebar are easy to see but the road image is still really dark while driving on a bright day

VirtuallyChris, 2021-04-20:
hmm you're right, would still be a huge help for all the devices with varying light sensor calibration.

ShaneSmiskol, 2021-04-21:
Yes! I assume from the road-facing camera? In traffic at night there's always a raised truck with full beams on making the device go full bright for no reason

closed 2021-05-26 by adeebshihadeh