commaai/openpilot #20724: Chrysler Pacifica 2017+ LKAS fault When Stopping/Starting with OP Engaged

ZeroAlligator, 2021-04-21:
Describe the bug
 When OpenPilot is engaged and the vehicle stops for traffic, OpenPilot sometimes fails to start driving from a stopped position, and the following error occurs: "LKAS fault: Restart car to engage". The following steps will always result in this fault, but there are other various situations where a LKAS fault occurs.

 Vehicle: Chrysler Pacifica 2017 (Non-Hybrid)
 LKAS threshold: 9 m/h

How to reproduce or log data
1.Power up OpenPilot (OP)
2.Allow OP to come to a stop behind another vehicle.
3.Wait in a stopped position for an extended period of time (approximately 30 or more seconds).
4.OP will not move the vehicle forward once traffic begins moving.
5.Press gas pedal to move the vehicle forward.
6.LKAS error "LKAS fault: Restart car to engage" will appear on the OP screen.
7.The car will indicate that the LKAS system is unavailable.
8.OP functionality is unavailable.

Expected behavior
 OP should be able to engage and re-engage driving when the vehicle comes to a stop in traffic. This functionality of OP works without causing an LKAS fault when the vehicle only stops momentarily/very brief stops.

Device/Version information (please complete the following information):
•Device: Comma Two
•Dongle ID: 410c721d6d35265f

•Route: 410c721d6d35265f|2021-04-21--12-35-12--5
•Timestamp: Just After ~12:40
&
•Route: 410c721d6d35265f|2021-04-21--12-22-18--5
•Timestamp: Just After ~12:27

•OP/Device Version & Firmware:
 openpilot v0.8.3-release
 Git branch: release2
 Git Commit: 9a0006fa0a 
 Panda Firmware: 18cd8199fe2184f3
 OS Version: NEOS 16.2
•Car make/model: Chrysler Pacifica 2017

Additional context
The fixes to the previous Pacifica 2020+ LKAS issue (https://github.com/commaai/openpilot/issues/2110#issuecomment-824297336) seems to have fixed some of the LKAS issues for the 2017+ Pacifica (LKAS faults occur in fewer situations while driving), but the above steps still cause LKAS faults in the 2017+.

xps-genesis, 2021-04-26:
this is caused by cancel button spamming on gas press which is broken. should be fixed with this change

https://github.com/commaai/openpilot/pull/20693

adeebshihadeh, 2021-04-29:
Does #20693 fix the issue?

adeebshihadeh, 2022-01-19:
@jyoung8607 is this expected to be fixed as well?

jyoung8607, 2022-01-19:
Yes.

The issue from the opening comment is the same as #23398. We no longer have the Chrysler gas press exception and therefore won't get into controls mismatch. Sucks that they've lost engaged-gas-resume for now, but #23538 will get them back to status quo ante, and if @adhintz's work comes together, they'll be even better off than before.

I can't reconcile the later comment from @whitecapri92 with the original issue, but the other fix from #20693 just missed the cut for 0.8.4 which he was running at the time. They are probably fine now. If not, they should open a new issue with more actionable details and logs.

closed 2022-01-19 by adeebshihadeh