commaai/openpilot #20736: Stuck in "registering device"

wasserbrunner, 2021-04-23:
Describe the bug

I had constant model lagging error, so I reinstalled openpilot. It got stuck in "registering device". I have tried factory reseting and reflashing, neither worked. I am still stuck in "registering device". I have tried multiple wifi connections. I've tried plugged into a computer and the wall with a strong charger.

How to reproduce or log data

I reproduce it by rebooting my device.

Expected behavior

Normal boot

Device/Version information (please complete the following information):

Device: Comma 2
Dongle ID: can't get
Version: Release2-staging
Car make/model: 2020 corolla.

Thank you for any help.

adeebshihadeh, 2021-04-23:
This may be related to a recent change on our backend. How did you wipe your device? Did you touch /persist at all? Can you get your device's serial number (SSH in and run `getprop ro.serialno`)?

wasserbrunner, 2021-04-23:
I'm not familiar with most of this kind of technical stuff. I don't know if I touched /persist. I just followed the directions for reflashing on a mac https://github.com/commaai/eon-neos

I don't know if I can ssh in because I reflashed it and can't setup ssh in the settings (cause it won't  boot). When I get home I'll see if workbench detects it

adeebshihadeh, 2021-04-23:
We pushed a change to our backend. If you connect your device to wifi and try again, it should work. Let us know how if you have any other issues!

wasserbrunner, 2021-04-26:
That worked thank you!!

ReFil, 2021-04-26:
I encountered this bug a few hours ago, using a FrEON, put in openpilot.comma.ai as the install link. reflashed twice to no avail

adeebshihadeh, 2021-04-26:
That's expected for freons on current release. You'll need to run master until the next release.

closed 2021-04-26 by adeebshihadeh