commaai/openpilot #20739: Toyota 2021 CAMRY HY LE Korea Version - Car Unrecognized: Dashcam Mode

ktaejung01, 2021-04-24:
**Describe the bug**

I just install OP 0.8.3 at my LePro3 Phone + BlackPanda.
And hook up every cable and module but car unrecognized dashcam mode.

I tried Fingerprint ver 1.0 via SSH = fail = may no more support
Also Fingerprint ver 2.0 at WIKI = still do not know how to do

**How to reproduce or log data**

**Expected behavior**

**Device/Version information (please complete the following information):**
 - Device: EON
 - Dongle ID:  d4f7a09c0931ccdd
 - Route: d4f7a09c0931ccdd
  - Timestamp: 1 Hour into the drive
 - Version: 0.8.3 - release 2
 - Car make/model:  Toyota / Camry HYB LE 2021 for Korea version

**Additional context**

ktaejung01, 2021-04-24:
Also I try finger print ver 2.0 but stuck at below car name = MOCK
Why not toyota ??

[image]

ktaejung01, 2021-04-24:
[image]
36: 8, 37: 8, 166: 8, 170: 8, 180: 8, 295: 8, 296: 8, 426: 6, 452: 8, 466: 8, 467: 8, 550: 8, 552: 4, 560: 7, 562: 6, 581: 5, 608: 8, 610: 8, 643: 7, 713: 8, 728: 8, 740: 5, 761: 8, 764: 8, 800: 8, 810: 2, 812: 8, 818: 8, 824: 8, 829: 2, 830: 7, 835: 8, 836: 8, 865: 8, 869: 7, 870: 7, 871: 2, 888: 8, 889: 8, 896: 8, 898: 8, 900: 6, 902: 6, 905: 8, 913: 8, 918: 8, 921: 8, 933: 8, 934: 8, 935: 8, 944: 8, 945: 8, 950: 8, 951: 8, 953: 8, 955: 8, 956: 8, 971: 7, 975: 5, 993: 8, 998: 5, 999: 7, 1000: 8, 1001: 8, 1002: 8, 1017: 8, 1020: 8, 1041: 8, 1042: 8, 1044: 8, 1056: 8, 1057: 8, 1059: 1, 1071: 8, 1076: 8, 1077: 8, 1114: 8, 1161: 8, 1162: 8, 1163: 8, 1164: 8, 1165: 8, 1166: 8, 1167: 8, 1228: 8, 1235: 8, 1264: 8, 1279: 8, 1541: 8, 1552: 8, 1553: 8, 1556: 8, 1557: 8, 1568: 8, 1570: 8, 1571: 8, 1572: 8, 1592: 8, 1593: 8, 1594: 8, 1595: 8, 1745: 8, 1779: 8, 1786: 8, 1787: 8, 1788: 8, 1789: 8, 1904: 8, 1912: 8, 1990: 8, 1998: 8

ktaejung01, 2021-04-24:
I just run "fw_versions.py" and still NO VIN & EMPTY FW version.

==================
root@localhost:/data/openpilot/selfdrive/car$ ./fw_versions.py
Getting vin...
vin query retry (1) ...
vin query retry (2) ...
vin query retry (3) ...
vin query retry (4) ...
vin query retry (5) ...
vin query retry (6) ...
vin query retry (7) ...
vin query retry (8) ...
vin query retry (9) ...
vin query retry (10) ...
VIN: 00000000000000000
Getting VIN took 1.083 s

100%|█████████████████████████████████████████████████████████████████████████████████████| 3/3 [00:02<00:00,  1.18it/s]

Found FW versions
{
}

Possible matches: set()
Getting fw took 2.642 s

ktaejung01, 2021-04-24:
====================
pandad.py
============
root@localhost:/data/openpilot/selfdrive$ ./pandad.py
Connecting to panda
Panda found, connecting
opening device 50001f001851393438333434 0xddcc
exception LIBUSB_ERROR_BUSY [-6]
Traceback (most recent call last):
  File "/data/pythonpath/panda/python/__init__.py", line 185, in connect
    self._handle.claimInterface(0)
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/usb1/__init__.py", line 1168, in claimInterface
    mayRaiseUSBError(
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/usb1/__init__.py", line 127, in mayRaiseUSBError
    __raiseUSBError(value)
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/usb1/__init__.py", line 119, in raiseUSBError
    raise __STATUS_TO_EXCEPTION_DICT.get(value, __USBError)(value)
usb1.USBErrorBusy: LIBUSB_ERROR_BUSY [-6]
connected
Panda None connected, version: v1.7.5-EON-7c00bc99-RELEASE, signature 18cd8199fe2184f3eb68762eb91b96928a9735e43bd1b6d9628c3adde53150a8e46e624e8d0c2a29aa469be661e350a0a7b5d38f2b0c2059d361fa41edf4497e56a58bf6167faf7f30ee9032a113ed2152e7f862ed1311c3b06616919d734c090e1551f4090c3157763be6081de35089799f216ddca337390df99d1069b34d4e, expected 18cd8199fe2184f3eb68762eb91b96928a9735e43bd1b6d9628c3adde53150a8e46e624e8d0c2a29aa469be661e350a0a7b5d38f2b0c2059d361fa41edf4497e56a58bf6167faf7f30ee9032a113ed2152e7f862ed1311c3b06616919d734c090e1551f4090c3157763be6081de35089799f216ddca337390df99d1069b34d4e
Resetting panda
r

crwusiz, 2021-04-26:
36: 8, 37: 8, 166: 8, 170: 8, 180: 8, 295: 8, 296: 8, 426: 6, 452: 8, 466: 8, 467: 8, 550: 8, 552: 4, 560: 7, 562: 6, 581: 5, 608: 8, 610: 8, 643: 7, 713: 8, 728: 8, 740: 5, 761: 8, 764: 8, 800: 8, 810: 2, 812: 8, 818: 8, 824: 8, 829: 2, 830: 7, 835: 8, 836: 8, 865: 8, 869: 7, 870: 7, 871: 2, 888: 8, 889: 8, 896: 8, 898: 8, 900: 6, 902: 6, 905: 8, 913: 8, 918: 8, 921: 8, 933: 8, 934: 8, 935: 8, 944: 8, 945: 8, 950: 8, 951: 8, 953: 8, 955: 8, 956: 8, 971: 7, 975: 5, 993: 8, 998: 5, 999: 7, 1000: 8, 1001: 8, 1002: 8, 1017: 8, 1020: 8, 1041: 8, 1042: 8, 1044: 8, 1056: 8, 1057: 8, 1059: 1, 1071: 8, 1076: 8, 1077: 8, 1114: 8, 1161: 8, 1162: 8, 1163: 8, 1164: 8, 1165: 8, 1166: 8, 1167: 8, 1228: 8, 1235: 8, 1264: 8, 1279: 8, 1541: 8, 1552: 8, 1553: 8, 1556: 8, 1557: 8, 1568: 8, 1570: 8, 1571: 8, 1572: 8, 1592: 8, 1593: 8, 1594: 8, 1595: 8, 1745: 8, 1779: 8, 1786: 8, 1787: 8, 1788: 8, 1789: 8

add toyota hybrid 2018 value

ktaejung01, 2021-04-26:
Special thanks for **_crwusiz,_** it is very helpful.

closed 2021-04-26 by ktaejung01