commaai/openpilot #20745: OP came to complete stop as lead car accelerated away

mattbsea, 2021-04-25:
**Describe the bug**

As I was approaching stopped cars at a light, OP correctly slowed down. As that happened the light turned green and the lead cars began to drive. OP came to a complete stop instead of following lead car.  Had to disengage and re-engage to resume drive. 

Looking at the HUD video again, op thought there was an object in front of me and was stopped for it.

**How to reproduce or log data**

https://my.comma.ai/9388d771f95175f5/1619301091233/1619301869326

Sat, 24 Apr, 2021 @ 14:56:16

**Expected behavior**

Correctly identify objects, ignore ghost objects

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 9388d771f95175f5
 - Route: 9388d771f95175f5|2021-04-24--14-51-31
 - Timestamp:  Sat, 24 Apr, 2021 @ 14:56:16
 - Version: 0.8.3 release2
 - Car make/model: Toyota RAV4H 2020

**Additional context**

pd0wm, 2021-04-29:
The car was detecting a false positive radar point right on the radar itself, probably due to the rain. Below a certain speed we don't require vision confirmation to stop anymore, in case you stop so close to another car it doesn't look like a car anymore. This cause the car to remain stationary because it thought there was a car right at your bumper.

These class of issues will be fixed when we switch to a vision only longitudinal policy.

[image]
(Radar point right on top of your own car)

mattbsea, 2021-04-29:
That makes sense.  Thanks for the explanation!

closed 2021-04-30 by pd0wm