commaai/openpilot #20746: Vehicle parameter identification failed

bowenh66, 2021-04-25:
2017 Toyota Corolla
[image]
After I reboot the system it works fine for a minutes until this message shows up.

pd0wm, 2021-04-25:
Can you fill out the bug report template? We need the logs to look into this.

 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

bowenh66, 2021-04-25:
Hi thanks for the reply!
I’m not too familiar with coding, is there anywhere I can get more
information regarding getting the log?

Thanks
Bowen

pd0wm, 2021-04-25:
Just fill out the template with all the required information (dongle id, route, version) and we can find the correct logs.

bowenh66, 2021-04-25:
- Device: comma two
   - Dongle ID: 308441ee81a066a4
   -
   - Route: my.comma.ai/useradmin I went on this link and this is what I
   see, I can’t find information for route number.
   - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the
   drive)]
   -
   - Version: release2 v0.8.3
   - Car make/model: 2017 Toyota Corolla

pd0wm, 2021-04-26:
Can you connect your device to wifi to make sure the full logs are uploaded?

pd0wm, 2021-04-26:
Looking at the low resolution logs (qlogs), it seems like the angle sensor on your steering wheel indicates ~10 degrees when going straight, which is outside of the acceptable range for openpilot to work. Does your steering wheel look straight when driving straight?

bowenh66, 2021-04-26:
That might be the exact problem, it might be an alignment issue. Or because
the car was rebuilt, it was hit at the rear frame but it’s drives pretty
straight.
My steering does appear to be towards the left when it’s driving straight,
and open pilot does an awesome job keeping the vehicle right in the center
of the lane.

When the Error Message comes up, if I disengage, it will not engage again
without rebooting. But if I keep it engaged with the error message, it
drives perfect.

Is there any way to bypass that acceptable range for the angle sensor
boundaries?

bowenh66, 2021-04-27:
Is there anything i can do to help on my end?

Thanks
Bowen

pd0wm, 2021-04-28:
The 10 degree limit is here: https://github.com/commaai/openpilot/blob/master/selfdrive/locationd/paramsd.py#L141

But there might be other subtle issues due to the asymmetry in your car. I would recommend to get it aligned.

closed 2021-04-28 by pd0wm

bowenh66, 2021-04-28:
Thank you very much!
How do I access those coding and change that limit on my device?