commaai/openpilot #20756: Constant model lagging error

wasserbrunner, 2021-04-26:
**Describe the bug**

The car can't engage OP for more than a couple of seconds without model lag error. It often can't engage at all. Also (seemingly unrelated..) sometimes upon bootup it gets stuck in "registering device...". This is _despite_ the recent fix, which did fit this issue for me but only sometimes. 

**How to reproduce or log data**

I just turn on my car and attempt to engage.

**Expected behavior**

OP boots up, and can engage.

**Device/Version information (please complete the following information):**
 - Device: C2
 - Dongle ID:  62b2d04d36644847
 - Route: 
 - Timestamp: Anytime in the last 2 weeks
 - Version: 0.8.3
 - Car make/model: 2020 corolla 

**Additional context**

Thank you!

adeebshihadeh, 2021-04-27:
Can you try re-flashing your device? See [this repo](https://github.com/commaai/eon-neos) for instructions. Is there anything else weird going on with your device?

VirtuallyChris, 2021-04-27:
Please reach out to support@comma.ai and we can help you with this.

wasserbrunner, 2021-04-27:
I actually have reflashed it to no avail. I’ll reach out to support thank you

closed 2021-04-28 by adeebshihadeh