commaai/openpilot #20760: 2017 Jeep cherokee KL try fix  car unrecognized -Fingerprinting-

kellersing, 2021-04-27:
i tired and trying to learn something else to fixing this edit, about how and why car unrecognized, our  2017 Jeep Cherokee has ACC, KLS or front camera forward ,   i think something maybe it would fix by fingerprinting if support to .     

[image]

i am used window Powershell via SSH wifi and location to /data/openpilot/selfdrive/car/Chryler/values.py
nano values.py   >Edit>Save>Exit

i change edit values.py, please check this below picture,  if i am doing correct or what?

[image]

but the problem is,  after we Save and Exit Edit because it need require to reboot device however it already restore itself values lost of my edit,  ,     i dont quit understand about powershell or was it device self-heal, 

**Device/Version information (please complete the following information):**
 - Device: Comma Two Pink
 - Dongle ID: e82e1bae71399c09
 - Route: e82e1bae71399c09|2021-04-26--17-20-00
 - Version: 0.8.3

kellersing, 2021-04-27:
@pd0wm would you check this for it?

VirtuallyChris, 2021-04-27:
Jeep doesn't use FP 2.0 yet. Please get your fingerprint the old way and add it as a subset of the existing fingerprint.

kellersing, 2021-04-27:
@VirtuallyChris  ok thank, i would try, on fingerprinting 1.0,   it a bit hard for me understand

VirtuallyChris, 2021-04-28:
oh just noticed it's not a Grand Cherokee so it's not supported yet and will likely need a new car port. I'll close this issue for now.

closed 2021-04-28 by VirtuallyChris