commaai/openpilot #20766: 2020 Corolla isn't fingerprinting all the required ECUs (only getting esp eps and camera)

mjaverto, 2021-04-28:
**Describe the bug**

Corolla isn't fingerprinting all the required ECUs. It is only getting esp eps and camera. (see Comma Support ticket #12479)

**How to reproduce or log data**

Turn on car with _new_ Comma 2 unit (see additional context below with why _new_ unit). When car is turned on, Comma turns on like normal and goes through boot sequence but car throws LTA errors, Cruise Errors. When unplugging comma and plugging back in stock connectors, errors go away.

Plug in Comma unit instead of stock and use it to drive.

Reproduce errors with Comma, and no errors with Stock
https://photos.app.goo.gl/rZY9tJaRH2a3jhmCA

Specific Errors: 
https://photos.app.goo.gl/yK7B8apy6j2ciVyWA

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: c0f760f08b83b7ba
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: Startup
 - Version: 9a0006fa0a / release 2 / v0.8.3
 - Car make/model: 2020 Toyota Corolla

**Additional context**

I ordered my comma last march, #7683.

Its been working well in a 2020 Corolla but this past week, as I got into my car I saw something about an upgrade, Neo I think?

I let the upgrade finish and all went well. I drove twice that day with no issues.

Two days ago, I was driving when all of a sudden I received a light saying "Malfunction with cruise control" or something to that effect from the Corolla; don't quote me on exact language. I noticed that the comma restarted, mid-driving. Once it turned back on and I power cycled the car, it worked again as normal. This was about 10m into a drive and it definitely wasn't hot outside, 40s, 50s maybe? Unsure what caused a reboot only 10m in.

Then later that day (about 60m after the reboot incident), I went to drive and the Comma2 won't turn on at all. No display. It looks like it has no power.

After browsing all of github issues, I can't seem to find anything about the comma not powering up at all. I did however learn there was a small light on the back near the power cord.

When power is plugged in but the car isn't on, I get a flashing blue light on the back. When the car is on, it changes to solid blue. However the display of the comma is off, nothing is happening still. The unit looks dead other than this light.

I've tried to hold power, hold power + volume up, power + volume down... Nothing. I of course pulled the power plug from the back of the unit dozens of times now. Pulled power and let it sit with no power all night. Tried to power again today, nothing. And when I saw power, I've always used the car so I've ruled out problems with amperage, right cord, etc from inside. (However, I can power with 2.4a Usb-a to usb-c outside of the car, I just get the blue blinking light similar to when plugged in within the car, with the car off.

I haven't done anything different in months, it's been working flawlessly. The only thing I can point to is that upgrade on Wednesday of this past week.

pd0wm, 2021-04-29:
Do you get errors with just the harness block plugged in (no comma two)?

mjaverto, 2021-04-29:
@pd0wm no errors when USB-C is unplugged from harness. It's like a stock corolla then with only harness plugged in.

mjaverto, 2021-05-04:
I was able to prove today this was a hardware issue. After replacing the USB-C cable with an extra I had just on a hunch, everything functioned as normal.

closed 2021-05-04 by mjaverto