commaai/openpilot #20798: Unexpected "vehicle parameter identification failed" alert

jyoung8607, 2021-05-02:
**Describe the bug**

While driving with openpilot engaged, relatively straight on an interstate highway, my device suddenly threw a VPID alert. My vehicle doesn't have any alignment or other mechanical issues that I'm aware of.

**How to reproduce or log data**

Unknown; this has never happened on my vehicle before.

**Expected behavior**

Don't throw VPID alerts for an otherwise working car/drive.

**Device/Version information (please complete the following information):**
 - Dongle ID: cae14e88932eb364
 - Route: cae14e88932eb364|2021-05-01--13-53-49
 - Timestamp: partway through segment 11
 - Version: `jyoung8607/mqb-fpv2-temp`, which is `commaai/master` as of 228308b719a5e1fb24ce912eeaa13eaf2db0a22e with some car port tweaks
 - Car make/model: 2018 Volkswagen Golf R

**Additional context**

liveParameters learned steerRatio and stiffness seem to jump straight from reasonable-looking to unpopulated.

Before (as far down in segment 11 as useradmin will show me):
```
( logMonoTime = 1294438288556,
  liveParameters = (
    valid = true,
    gyroBias = 0,
    angleOffsetDeg = -0.50703567,
    angleOffsetAverageDeg = -0.569617,
    stiffnessFactor = 0.99630338,
    steerRatio = 15.618905,
    sensorValid = true,
    yawRate = 0,
    posenetSpeed = 0,
    posenetValid = true ),
  valid = true )
```

After (first logged event in segment 12):
```
( logMonoTime = 1344437892359,
  liveParameters = (
    valid = false,
    gyroBias = 0,
    angleOffsetDeg = nan,
    angleOffsetAverageDeg = nan,
    stiffnessFactor = nan,
    steerRatio = nan,
    sensorValid = true,
    yawRate = 0,
    posenetSpeed = 0,
    posenetValid = true ),
  valid = true )
```

cferra, 2021-05-03:
Same situation here with a 2018 wrx

https://my.comma.ai/cabana/?route=d345d482f9119a3f%7C2021-05-02--15-56-59&exp=1651537430&sig=4ikuEGQgcju07zSE59X3QW9lLi5Z%2FStz%2Ftd2zjbvmSI%3D&max=17&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2Fd345d482f9119a3f%2F826d173db65a8c1063592edf7dc56a80_2021-05-02--15-56-59

HaraldSchafer, 2021-05-03:
Seems like there it was possible for the localizer to trigger some kind of numerical instability somewhere since the 100Hz localizer change. 

I reverted those changes, while we figure out what's happening and how to fix it. We'll fix it asap and ensure that nans in the localizers can't cause the param learner to get nans too.

HaraldSchafer, 2021-05-06:
Fixed with https://github.com/commaai/openpilot/commit/e9db5723ef348954118643501a92cf0715402fea

closed 2021-05-06 by HaraldSchafer