commaai/openpilot #20832: Peugeot e2008

itrack, 2021-05-05:
Hi!
Plans to integrate PSA cars in Openpilot? Like Peugeot e2008, Peugeot e208, Opel Corsa-e?

VirtuallyChris, 2021-05-05:
That car likely doesn't have LKAS or ACC which is needed for openpilot. Plus, this isn't where you would ask to add support. All support comes from community contributions, so feel free to make a PR if you can get it working on your car!

closed 2021-05-05 by VirtuallyChris

itrack, 2021-05-06:
But cars have LKAS and ACC, the cars is autonomous level 2

dorkmo, 2021-05-06:
@itrack its hard to know for sure if a new vehicle will work will openpilot, you might start exploring the car's stock hardware. you might try to find the camera and its plugs. post pictures of what you find. it may share parts with other cars.

VirtuallyChris, 2021-05-17:
Great if it has LKAS and ACC you will need to do a brand port. Theres some helpful info on some old blog posts on blog.comma.ai. Otherwise, the community needs to add support on their own. We have an #other-cars channel on discord.comma.ai if you want to ask for help from others.