commaai/openpilot #20841: Problem of touch issues.

johnlu6518, 2021-05-06:
The options for reboot and turn off are difficult to click on the screen. 
(The touch is not smooth, and the display cannot be moved up and down)

 Device:comma two
 Dongle ID: ec30903bb15db3ad

adeebshihadeh, 2021-05-06:
Can you be more specific? It's the normal android power menu, which doesn't move.

johnlu6518, 2021-05-07:
Ok，I will take photos or video for u later.

johnlu6518, 2021-05-07:
https://youtube.com/shorts/f_BMZKt-2iU?feature=share

VirtuallyChris, 2021-05-07:
From the video it looks like you tried to scroll from the bottom of the screen up; but you can only scroll within the box with the actual settings in it. If you scroll with your finger in the middle of the screen to start, are there any issues?

johnlu6518, 2021-05-09:
Yes, you are right. Scroll with finger in the middle of the screen.