commaai/openpilot #20853: Do not issue Driver distracted when standing still

yegor-mialyk, 2021-05-07:
Standing still behind a lead car at an intersection, watching for crossing traffic and pedestrians. OP issues a Driver distracted event cause my eyes are pointed to the left (or right) direction. While this event (EventName.preDriverDistracted?) is not supposed to initiate an audible alert, the VisualAlert.steerRequired in Hyundai (Santa Fe 2020) will beep at you anyway.

I would suggest not to issue Driver distracted when standing still.

jyoung8607, 2021-05-07:
I think the design on openpilot's display is deliberate, and IMO correct (visual only while at standstill, escalating if moving). Insofar as it also triggers `HUDControl.VisualAlert.steerRequired`, I do agree with you.

IMO the same thing applies with auto lane-change. It's not something that should trigger driver takeover alerts in the instrument cluster.

adeebshihadeh, 2021-05-07:
This is by design, however the alert shown on your dash happens to also play a sound. We'd certainly merge a PR if you found a message to show the dash alert without the sound.

closed 2021-05-07 by adeebshihadeh