commaai/openpilot #20872: Can error appears while driving

johnlu6518, 2021-05-11:
**Describe the bug**

Suddenly an error message appears while driving, and it can work after reboot.

Device: Comma 2
Dongle ID: ec30903bb15db3ad
Route: ec30903bb15db3ad|2021-05-10--20-37-49
Timestamp: 4 min into the drive
Version: 0.8.3
Car make/model: Lexus IS 200T (2017)

pd0wm, 2021-05-11:
The raw logs for that drive are not uploaded. Make sure the "Upload Raw Logs" toggle is on, and your device is connected to wifi.

johnlu6518, 2021-05-11:
But I am not sure where is the toggle for Upload Raw Logs.. Can you help me find it?

johnlu6518, 2021-05-11:
It's midnight now, I will keep the device on all night without turning off for u check, Thank you!

pd0wm, 2021-05-11:
Everything is uploaded now. I'll check it out tomorrow.

johnlu6518, 2021-05-12:
It happened again today, lost Panda and can't use it anymore.
[image]

pd0wm, 2021-05-12:
This is a known bug when the modem tries to switch cell towers and uses too much power. Some devices seem to be more affected than others. We use https://github.com/commaai/openpilot/issues/19874 to track this issue.

You can do two things:
 - Remove your sim card for the time being. We're working on a software fix, which is expected in 0.8.5, but it could be longer.
 - If your device is still under the 1 year warranty, contact support@comma.ai and we can replace it for you.

johnlu6518, 2021-05-12:
After removing the sim card, what are the restrictions and differences in use?  Does it work properly?

pd0wm, 2021-05-12:
No difference in driving. Just make sure you connect to wifi to get updates.

johnlu6518, 2021-05-12:
Okay, Thank you!

nabeelr, 2021-06-01:
Just adding some details from my other now closed ticket (#21057) into this one for reference:

**Describe the bug**

Occasionally, my C2 will say "No Panda" along with a CAN error message. The car (a 2019 VW Golf R) will also throw a bunch of errors at that time with a bunch of modules. A reboot is needed to correct the issue, on the C2. Sometimes a restart of the car is needed to clear it's issues too. The C2's panda will remain offline without rebooting the C2, despite what you do to the car.

**How to reproduce or log data**

Unable to reliably reproduce. Driving data exists however seems cut short.
Issue occurred at the end of the drive on this segment 903210ecc0cb706b|2021-05-25--21-13-22--80
OP was engaged right up until the end of the drive, despite Explorer saying it was disengaged.

**Expected behavior**

For Panda not to go offline randomly.

**Device/Version information (please complete the following information):**
 - Device: Comma2
 - Dongle ID: 903210ecc0cb706b
 - Route: 903210ecc0cb706b|2021-05-25--21-13-22--80
 - Timestamp: 22:34, right at the end of the drive. After the footage ended, about 10 seconds later in that roundabout, OP disengaged, showed the red "CAN Error" message and disengage warning, the car's LKA and ACC went offline, and OP screamed until I rebooted it about 30 seconds later.
 - Version: 0.8.4
 - Car make/model: 2019 Golf R Manual

**Additional context**

Lots of errors logged in the cars ABS, LKA, ECU, EPS, and BCM modules after this this issue. These errors do not occur without the C2 present.

When I first got my C2, I did have an issue with tower changes causing issues, but that has since been fixed after going from 0.7.4 to 0.7.8 (I think?). I used to be able to reproduce that issue by driving on a specific road, however driving that road doesn't cause this issue and hasn't caused any issues in 8 months.

My C2 DID hard lock a couple times in the winter (#19874), but after re-flashing NEOS it seems to have stopped doing that. This panda issue is fairly new, only in the last couple weeks has it done it (Maybe since 0.8.4? but it may have happened on 0.8.3 too. My memory is shit sooo...), and it's done it twice so far. The first time I ignored, and only took note the second time it happened. 

Update 1:

It happened again today, BUT there were also two other issues that occurred after a reboot shortly after in the same area. Though these presented in a new way that I don't think I've seen happen before. Not sure if they are related or not, but I can go back and test again to see if they happen in that area again.

The first thing that went wrong was the issue I described earlier, CAN connection error, red warning screen, No Panda afterwards, though the panda firmware still showed up in the Developer tab.
The only way I was able to correct this issue was with a reboot of the C2.
This DID occur in an area with poor cell reception, and the C2 was constantly saying that it was gaining and losing cell service. Though the other times it has happened, weren't in areas of poor cell service, and never seemed to be reproducible before even on a road where sell service had caused previous issues. Maybe it will be on this road? I can try again tomorrow.
This occurred at: 903210ecc0cb706b|2021-05-29--17-04-20--30 at approximately 17:35:01
[image]
[image]

Later that same drive (though after rebooting the C2), I saw it do the two other things I mentioned in: 903210ecc0cb706b|2021-05-29--17-46-50--5 at approximately 15:52:31 I got an error that I wasn't able to read, but I suspect it was another CAN Error. Panda stayed online, but later trying to re-engage showed an error saying something about controls mismatch somewhere after 17:52:56.

Approximately at 17:53:50, I turned the ignition off and right back on again (not giving the car enough time to reset or for modules to go to sleep) and was able to drive again with my C2 without incident for the rest of the time back home.

During the ride back home, after the reboot of the C2 I kept an eye on the "connected" status and cellular signal strength of the C2 and saw it go from red to yellow to white several times, without issue or incident. I also observed the C2 gain and lose cell service, switch back and forth between 2G, 3G and LTE without issue.

I'll probably do this drive again tomorrow, let me know if you want me to try it again, and I can make it a priority.

Hope this helps!

Update 2:

So, I managed to find a stretch of road where I could make the issue occur consistently and compared with and without the sim card. which definitely seemed to cause the issue. At one point, the C2 lost the panda entirely, and I had to disconnect power from the C2 entirely for it to come up again. When this happened the Panda firmware section was blank.

I did the road 3 times each, up, back down, and up again.

Here are the drives:
903210ecc0cb706b|2021-05-30--22-40-33--27
903210ecc0cb706b|2021-05-30--23-09-52--0
903210ecc0cb706b|2021-05-30--23-14-29--0

Without the SIM I was able to do all 3 drives together, then popped the sim back in for a 4th run, where it died :
903210ecc0cb706b|2021-05-30--23-19-40--0

The drive with the SIM card starts at approximately 23:32:50, and ends at the end of the clip.

The problem manifested in many different ways. From the panda going offline, to the C2 thinking the car was off, to getting a controls mismatch error, or a CAN communication error.

I can very consistently reproduce this issue, so if you guys want me to do some testing on that road, I'll be more than glad to help out. ;)

Positev, 2021-06-02:
> Just adding some details from my other now closed ticket (#21057) into this one for reference:
>
> **Describe the bug**
>
> Occasionally, my C2 will say "No Panda" along with a CAN error message. The car [...]

I have the exact same issue. I suppose i should wait to return my C2? Any temporary solution to this?

nabeelr, 2021-06-07:
Hey all! I just wanted to offer an update, today on master-ci, I went for a drive in a remote area, and I had some similar issues to this when there was poor service again. I had heard that there was hopes that 0.8.5 would fix this issue.

It presented a bit differently, it showed a USB error, and then I lost the Panda. After a few seconds, OP acted like I had shut the car off, and the only way I could get the Panda back was if I unplugged my C2 from power, and plugged it back in. Powering off, or rebooting wouldn't solve the issue. This happened a couple times on this stretch.

I was able to reproduce it on this drive: 903210ecc0cb706b|2021-06-06--17-09-48--1 at around 17:13:55

I was running the master-ci branch of openpilot-v0.8.5-f04d6a9b3-2021-06-5T and NEOS 17.

I hope this helps. I can do more testing if needed. Let me know how I can help. :)

adeebshihadeh, 2021-06-07:
Thanks for the update @nabeelr. The fix unfortunately didn't make it in for 0.8.5, but we're able to reproduce exactly what you're seeing. The USB error alert was added to catch this. The best way to help right now is preserve your route with the issue and post them here to help us confirm that the fix will cover your case.

nabeelr, 2021-06-07:
Cool, will do! Thanks so much.

adeebshihadeh, 2021-06-23:
We have a better understanding of this issue now, and I added some logging around the cell connection so we can target the fix better. Would either of you guys be able to switch to master-ci and upload some routes?

nabeelr, 2021-06-23:
@adeebshihadeh Will do!

I'll be going for a drive later tonight, and I'll do so running master CI.

nabeelr, 2021-06-24:
Got it to happen, and captured it on master-ci from today at about 14:00 EDT

On drive: 903210ecc0cb706b|2021-06-23--17-49-34--27
@ timestamp 18:17:20

Need me to pull logs in some other way? Let me know if you need more info.

Smartype, 2021-06-26:
Whenever usb drops, Boardd cleans ControlsIsReady param, re-connect to panda, and then wait for controlsd to set ControlsIsReady, which will never happen. 
You will have to simulate an offload event to reload controlsd, or simply reboot the devices.

nabeelr, 2021-06-30:
@adeebshihadeh 

So, I recently tried that route again, A/B testing my older C2 that has this issue, vs my newer C2, but I couldn't get either to do it...

The one variable is that I drove there (about 40 minutes out) on my newer C2, and swapped to my older C2 after doing the run with my newer C2 without issue. I swapped the SIM card into whichever C2 I was using at the time, of course...

But it made me think, could it be temperature dependent? It seems weird that the problem suddenly fixed itself on my older C2, and the only variable was that it had been off for the 40 minute drive to the location...

Maybe I'm grasping at straws? IDK What are your thoughts? Now that I have two C2s, I can do some A/B testing if we think it's hardware dependent (which I gather we think it is because comma was offering an in warranty exchange for them AFAIK?) My older one is still in warranty, but I kinda want to see this problem to resolution instead of just replacing it, unless we think there won't be a software fix possible.

adeebshihadeh, 2021-07-06:
Thanks for the testing and logs @nabeelr! Logging for this issue has been in release for over a month now, and we've only seen this issue on a handful of devices. The issue seems to be isolated to particular bands on particular devices, which is why it's local to certain areas. If you have an affected device, you can email support@comma.ai for a replacement.

closed 2021-07-06 by adeebshihadeh