commaai/openpilot #20885: Cannot build openpilot tools with scons in Ubuntu 18

petrovicu, 2021-05-12:
**Describe the bug**

When following the guide to build openpilot for Ubuntu, I stucked with: "scons -j$(nproc)". The error message:

scons: Reading SConscript files ...
AttributeError: _RSAobj object has no 'size_in_bits' attribute:
  File "/home/wingman2/openpilot/SConstruct", line 392:
    SConscript(['panda/board/SConscript'])
  File "/home/wingman2/.local/lib/python3.6/site-packages/SCons/Script/SConscript.py", line 654:
    return method(*args, **kw)
  File "/home/wingman2/.local/lib/python3.6/site-packages/SCons/Script/SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/wingman2/.local/lib/python3.6/site-packages/SCons/Script/SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/wingman2/openpilot/panda/board/SConscript", line 144:
    certs = [get_key_header(n) for n in ["debug", "release"]]
  File "/home/wingman2/openpilot/panda/board/SConscript", line 144:
    certs = [get_key_header(n) for n in ["debug", "release"]]
  File "/home/wingman2/openpilot/panda/board/SConscript", line 67:
    assert(rsa.size_in_bits() == 1024)
  File "/usr/lib/python3/dist-packages/Crypto/PublicKey/RSA.py", line 126:
    raise AttributeError("%s object has no %r attribute" % (self.__class__.__name__, attrname,))

By looking at openpilot/panda/board/SConscript script line 65, I can see that the script is looking for existing .pub files, but for some reason python3 Crypto lib is not able to get a 'size_in_bits' attribute.

**How to reproduce or log data**

Just follow the README instructions from:
https://github.com/commaai/openpilot/tree/master/tools
as well as:
https://youtu.be/ixfAdv9sL30?t=421

**Expected behavior**

openpilot tools builds successfully.

**Additional context**

OS: Ubuntu 18.x
Running scons through the python 3 pip

pd0wm, 2021-05-12:
You need to use the `pycryptodome` package instead of `pycrypto`. Did you install all dependencies from the pipfile?

petrovicu, 2021-05-12:
According to build log I did:
Installing dependencies from Pipfile.lock (448fa1)…
  🐍   ▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉ 0/0 — 00:00:00

----   FINISH OPENPILOT SETUP   ----
Configure your active shell env by running:
source ~/.bashrc

petrovicu, 2021-05-12:
I decomposed ubuntu_setup.sh script and run all commands manually. The previous problem has gone, but now I have a new issue:

/usr/local/lib/libzmq.a(src_libzmq_la-ctx.o): In function `zmq::mutex_t::mutex_t()':
ctx.cpp:(.text._ZN3zmq7mutex_tC2Ev[_ZN3zmq7mutex_tC5Ev]+0x27): undefined reference to `pthread_mutexattr_init'
ctx.cpp:(.text._ZN3zmq7mutex_tC2Ev[_ZN3zmq7mutex_tC5Ev]+0xa4): undefined reference to `pthread_mutexattr_settype'
/usr/local/lib/libzmq.a(src_libzmq_la-ctx.o): In function `zmq::mutex_t::~mutex_t()':
ctx.cpp:(.text._ZN3zmq7mutex_tD2Ev[_ZN3zmq7mutex_tD5Ev]+0x9b): undefined reference to `pthread_mutexattr_destroy'
/usr/local/lib/libzmq.a(src_libzmq_la-pipe.o): In function `zmq::mutex_t::try_lock()':
pipe.cpp:(.text._ZN3zmq7mutex_t8try_lockEv[_ZN3zmq7mutex_t8try_lockEv]+0x23): undefined reference to `pthread_mutex_trylock'
/usr/local/lib/libzmq.a(src_libzmq_la-thread.o): In function `thread_routine':
thread.cpp:(.text+0xc6): undefined reference to `pthread_sigmask'
/usr/local/lib/libzmq.a(src_libzmq_la-thread.o): In function `zmq::thread_t::start(void (*)(void*), void*)':
thread.cpp:(.text+0x1ec): undefined reference to `pthread_create'
/usr/local/lib/libzmq.a(src_libzmq_la-thread.o): In function `zmq::thread_t::stop()':
thread.cpp:(.text+0x29c): undefined reference to `pthread_join'
/usr/local/lib/libzmq.a(src_libzmq_la-thread.o): In function `zmq::thread_t::applySchedulingParameters()':
thread.cpp:(.text+0x72a): undefined reference to `pthread_setaffinity_np'
/usr/local/lib/libzmq.a(src_libzmq_la-thread.o): In function `zmq::thread_t::setThreadName(char const*)':
thread.cpp:(.text+0x7ea): undefined reference to `pthread_setname_np'
compile_dbc(["opendbc/can/dbc_out/cadillac_ct6_object.cc"], ["opendbc/cadillac_ct6_object.dbc", "opendbc/can/dbc_template.cc"])
clang: error: linker command failed with exit code 1 (use -v to see invocation)
scons: *** [cereal/messaging/bridge] Error 1
scons: building terminated because of errors.

pd0wm, 2021-05-12:
This is not really the place to troubleshoot issues with non-supported operating system. If there is something wrong with the Dockerfile or the ubuntu setup script on a fresh 20.04 install you can open a new issue.

This is better suited for #dev-openpilot on our discord (discord.comma.ai).

closed 2021-05-12 by pd0wm