commaai/openpilot #20904: Stopping at red light OP get confused because of a motorbike

adam-nelson, 2021-05-14:
**Describe the bug**

OP wasnt able to detect a motorcyclist/motorbike correctly at set of traffic lights

**How to reproduce or log data**

Looking at segment 18e8ac91b805db11|2021-05-14--19-09-11--50
- OP is slowing the car down
- For second the radar detecting another object and stop tracking the bike
- Car on engaged doesnt feel like it keeping a safe braking distance
- For some reason vision/radar has conflict because of a pole on the siding of the road behind the traffic light

**Expected behavior**

OP should be able to see the difference between undriveable object and exposed human riding a motorbike/motorcycle without human policy override at last second.

**Device/Version information (please complete the following information):**
 - Device: [comma two]
 - Dongle ID: [18e8ac91b805db11]
 - Route: [18e8ac91b805db11|2021-05-14--19-09-11--49]
 - Timestamp: [Fri, 14 May, 2021 @ 19:58:45]
 - Version: [release2]
 - Car make/model: Toyota corolla hybrid accent sport 2019 - australia delivered

**Additional context**

This google maps in the daylight. In Australia these signs are called Narrow marker signs. 
https://www.google.com.au/maps/@-27.3457235,153.0125702,3a,75y,122.51h,81.47t/data=!3m6!1e1!3m4!1sWBf7PQd3BRXuOZ_-huwAWg!2e0!7i16384!8i8192

adam-nelson, 2021-05-17:
- [ ] @pd0wm do you think this bug or something that could be resolved end to end vision planning is matured? And radar becomes optional?

If not a bug then close 😃

pd0wm, 2021-05-17:
Don't think openpilot got confused about the pole. The model placed the motorcycle much farther away, and fused it with the radar points from the cars in the intersection. It detect the motorcycle correctly around the time where you disengaged.

@HaraldSchafer any thoughts?

adam-nelson, 2021-05-17:
What seen in the gui/hud version. It was locking on something on the left like the pole

pd0wm, 2021-05-17:
You're probably referring to this. It picked up one of the cars crossing the intersection behind the pole.

[image]

HaraldSchafer, 2021-05-18:
For a vehicle to be correctly detected both these things need to be true:

1. The segnet needs to correctly identify the vehicle for all cases that look similar to this
2. The driving model needs to be able to see the vehicle is there if the ground truth is there

Both the segnet and driving model struggle in scenes like this. This will gradually improve as the models gradually improve, I don't think there's anything else we can do about it. We've begun working on revamping longitudinal policy, but tbh I think this case will be hard for a while.

closed 2021-05-19 by HaraldSchafer