commaai/openpilot #20909: Simulation stops for invalid vehicle parameters

siggie0815, 2021-05-14:
When starting latest OP simulation, few seconds after engaging OP it disengages with the following alert message: _"Vehicle Parameter Identification Failed"_

Also see this [discord conversation](https://discord.com/channels/469524606043160576/684457101455786017/841788306970247198).

The [proposed fix works](https://discord.com/channels/469524606043160576/684457101455786017/841953842991595570), but is obviously not intended to be used in real world use.

I also can confirm, the issue did not exist before May 6th.

creikey, 2021-05-16:
Ran a git bisect and fc3da5eaf9e9448825115e19dd59e1a2657ab249 seems to be the offender

creikey, 2021-05-16:
Looking at that commit, replacing `if (log.getStandstill())` L287 of locationd.cc with `if (this->car_speed < 1e-3)` fixes the issue on master

adeebshihadeh, 2021-05-16:
The sim doesn’t properly set the WHEELS_MOVING bit here https://github.com/commaai/openpilot/blob/master/tools/sim/lib/can.py#L42.

creikey, 2021-05-16:
> The sim doesn’t properly set the WHEELS_MOVING bit here https://github.com/commaai/openpilot/blob/master/tools/sim/lib/can.py#L42.

Just curious, where are these car can messages parsed into updates for the CarState capnp struct?

PR #20921 (2021-05-16): Set standstill bit in sim
 Fixes #20909

**Description**
https://github.com/commaai/openpilot/issues/20909
Sim didn't send that the wheels were moving over CAN causing openpilot to disengage in the simulator
**Verification**
Sim works
**Route**
Sim
closed 2021-05-16 by adeebshihadeh