commaai/openpilot #20912: Radar not publishing firmware version 2018 Honda fit ex

el11nene11, 2021-05-14:
[image]

Comma 2 states on screen car not supported. Support says radar isn't publishing firmware version so it's causing this.

VirtuallyChris, 2021-05-14:
Please add your Dongle ID and a route as stated in the template when you create a bug.

el11nene11, 2021-05-14:
> Please add your Dongle ID and a route as stated in the template when you create a bug.

Just edited

adeebshihadeh, 2021-06-13:
@el11nene11 is this still an issue? I don't see any recent routes from your device.

el11nene11, 2021-06-16:
No. This isn't an issue anymore. With the replacement comma two the issues were resolved. Thanks

closed 2021-06-16 by adeebshihadeh