commaai/openpilot #20914: Toyota: alert when speed is too low to engage

adeebshihadeh, 2021-05-15:
The cruise button states are available over CAN, so we should be able to detect attempts to engage.

Himanshunitrr, 2021-05-24:
I will like to work on this issue, can you elaborate whats the issue. (I am new to this codebase)

ntegan, 2021-05-25:
Some toyotas don't allow cruise/op to engage when under 25-28mph. 
I think this issue is saying it is possible to have op detect when driver attempts to engage op but is under this speed, and display a warning

primitivehacker, 2021-05-30:
What is CAN?

cydia2020, 2021-06-05:
I am not sure if this doable, but from what I have observed, at least on my Prius, CRUISE_STATE and CRUISE_CONTROL_STATE only changes when all conditions are met, and DSU_CRUISE doesn't give any useful information whatsoever.

However this might be different on other vehicles.

ahowe444, 2021-06-06:
> I am not sure if this doable, but from what I have observed, at least on my Prius, CRUISE_STATE and CRUISE_CONTROL_STATE only changes when all conditions are met, and DSU_CRUISE doesn't give any useful information whatsoever.
> 
> However this might be different on other vehicles.

Maybe we can check if vEgo is less than LOW_SPEED_LOCKOUT and if cruiseState.available is False, and then trigger the alert. Are there other reasons why cruise isn't available?

nanshao133, 2021-06-09:
中国版2018款凯美瑞混合动力更换了全速域雷达的，低速巡航踩刹车会警报ACC巡航不可用

cydia2020, 2021-06-09:
> 中国版2018款凯美瑞混合动力更换了全速域雷达的，低速巡航踩刹车会警报ACC巡航不可用

Probably unrelated to the topic of this issue, but are you talking about the chime/warning produced by the dash when openpilot emits pcm_cancel_cmd while the driver is engaging the brakes?

This issue is looking for a way of alerting the user when the speed is too low for openpilot to engage.

我沒太懂你說得什麼但是好像和這個issue關係不是很大。你說的是用戶踩剎車時如果openpilot發送取消信號導致車提示錯誤的問題嗎？

這個issue是說要在openpilot上顯示一個速度不夠engage的警告

nanshao133, 2021-06-09:
我遇到的问题是，第八代凯美瑞在启用acc激活op后，当车速低于45码，驾驶员踩刹车的时候，车仪表盘会出现ACC巡航不可用的提醒，并有噹的一声响声，几乎每次在op巡航过程中只要速度低于45码踩刹车，仪表盘都会提示，op上没有任何提示的

Smartype, 2021-06-10:
这是因为你踩刹车的时候op要退出acc。发送了cancel cmd. 而这个时候ecu其实还没有退出acc，有一个时间差。

这个时候，车身ecu接收cancel cmd，退出了acc，但是radar不知道，继续发送油门/刹车控制。ecu就报错了。

adeebshihadeh, 2021-11-08:
#22749 is a more generic version of this issue

closed 2021-11-08 by adeebshihadeh