commaai/openpilot #20915: Run openpilot with webcam on PC

LIHaoran-good, 2021-05-15:
I refer to this guideline that [Build openpilot for webcam](https://github.com/commaai/openpilot/blob/master/tools/webcam/README.md), when entering the compile instruction "USE_WEBCAM=1 scons -j$(nproc)",  report the error "scons: *** Import of non-existent variable "libkf" ", as shown below.

**Expected behavior**

Run openpilot with webcam on PC successfully

Operating system: [Ubuntu 20.04]

adeebshihadeh, 2021-05-15:
Are you able to build the non-webcam version of openpilot (i.e., without USE_WEBCAM=1)?

LIHaoran-good, 2021-05-15:
Thanks for your reply.
I am not able to build the non-webacam version of openpilot, when entering the compile instruction " scons -j$(nproc)",  still report the error "scons: *** Import of non-existent variable "libkf" ,too.
In addition, it is very strange that I only input  "scons", will also report this error, is it possible that there is a problem with the Scons installation ?  I don't know what  "libkf" is ,what does it do. 
What's the different between non-webacam version of openpilot and webacam version of openpilot, I think they are the same version of software.

adeebshihadeh, 2021-05-15:
The webcam flag changes the backend in camerad. libkf is from the rednose submodule, so make sure you have the submodules initialized and up to date. The issue tracker isn't the right place to get help on building openpilot. Try our community discord instead.

closed 2021-05-15 by adeebshihadeh

LIHaoran-good, 2021-05-17:
Could you please tell me how to initialized the submodules ? Thanks very much.