commaai/openpilot #20917: GENESIS G90 2019 Forward Collision-Avoidance Assist(FCA)

barghe, 2021-05-15:
**Describe the bug**

**How to reproduce or log data**

1. OpenPilot Cruise
2. set
3. Immediately after ingain, disgain
3. Warnings Occurred: Forward Collision-Avoidance Assist(FCA)

**Expected behavior**

**Device/Version information (please complete the following information):**
 - Device: EON/EON
 - Dongle ID: 
   1. 5d18df9ff3378701,
      Sat, May 15 14:38 to 14:52
      Sat, May 15 13:59 to 14:11
 - Route: route_name: 
   1. 5d18df9ff3378701|2021-05-15--05-38-50
   2. 5d18df9ff3378701|2021-05-15--13-59-15
  
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: release2 0.8.3
 - Car make/model: GENESIS G90 2019

**Additional context**

barghe, 2021-05-20:
CAR fingerprint GENESIS G90 2019
`67: 8, 127: 8, 304: 8, 320: 8, 339: 8, 356: 4, 358: 6, 359: 8, 544: 8, 549: 8, 593: 8, 608: 8, 688: 5, 809: 8, 832: 8, 854: 7, 870: 7, 871: 8, 872: 8, 897: 8, 902: 8, 903: 8, 905: 8, 909: 8, 916: 8, 1040: 8, 1042: 8, 1056: 8, 1057: 8, 1064: 8, 1078: 4, 1107: 5, 1113: 8, 1136: 8, 1141: 8, 1142: 8, 1143: 8, 1150: 4, 1151: 8, 1155: 8, 1156: 8, 1157: 4, 1162: 8, 1164: 8, 1168: 7, 1170: 8, 1173: 8, 1180: 8, 1184: 8, 1186: 2, 1191: 2, 1210: 8, 1265: 4, 1280: 1, 1281: 3, 1287: 4, 1290: 8, 1292: 8, 1294: 8, 1312: 8, 1322: 8, 1342: 6, 1345: 8, 1348: 8, 1363: 8, 1369: 8, 1371: 8, 1378: 8, 1384: 8, 1407: 8, 1419: 8, 1427: 6, 1434: 2, 1456: 4, 1470: 8, 2003: 8, 2004: 8, 2011: 8, 2012: 8`

ShaneSmiskol, 2021-06-02:
@barghe Can you make sure you upload the logs for the routes you listed? Make sure the upload raw toggle is enabled in settings. Visit https://my.comma.ai/useradmin/?onebox=5d18df9ff3378701%7C2021-05-15--13-59-15 and make sure the None's in the third column are replaced with log links

barghe, 2021-06-03:
On June 6th, we will activate the Raw Logs toggle switch and upload the video.

barghe, 2021-06-03:
> @barghe Can you make sure you upload the logs for the routes you listed? Make sure the upload raw toggle is enabled in settings. Visit https://my.comma.ai/useradmin/?onebox=5d18df9ff3378701%7C2021-05-15--13-59-15 and make sure the None's in the third column are replaced with log links

barghe, 2021-06-07:
[Raw Logs] Activated toggle switch and uploaded video.
My device is two.

- Device:
  1. ae415a196fa60e35
  2. 719b0f8c2f61284e

adeebshihadeh, 2021-06-19:
Neither of those routes are fully uploaded. Make sure to enable the toggle and wait until all four columns are populated in useradmin (i.e., none of the rows say 'None'). Also looks like we don't have a test route for the Genesis G90. Do you have a route you'd be willing to make public so we can use it for our public CI tests? The tests ensure there's no regressions for your car.

adeebshihadeh, 2021-06-23:
Were you able to get any routes fully uploaded?

barghe, 2021-06-23:
I will upload it again. June 25th.

barghe, 2021-06-29:
1. Activated [Raw Logs] toggle switch.

2. It is not populated in useradmin (i.e., no row marked 'none').

The HKG branch solved the problem.
It was modified by HKG Branch.
I succeeded in that.

I shall close this issue.

thank you.

closed 2021-07-08 by barghe