commaai/openpilot #20920: Failure to accelerate quickly after a cut-out

TKassis, 2021-05-15:
**Describe the bug**

OP fails to quickly accelerate when a slow car cuts out (moves to another lane). Had to press the gas pedal to accelerate as there were speeding cars behind me in the lane.

**How to reproduce or log data**

Have the lead car slow down considerably and switch lanes. OP will not accelerate quickly.

**Expected behavior**

OP should be able to detect a cut-out and begin acceleration much sooner.

**Device/Version information (please complete the following information):**
 - Device: Comma Two
 - Dongle ID: cd21d47a9d46bdf3
 - Route: cd21d47a9d46bdf3|2021-05-15--12-11-19--18
 - Timestamp: Mary 15 @ 12:29:32
 - Version: 0.8.3
 - Car make/model: Toyota Corolla Hybrid 2021

**Additional context**

[image]

adam-nelson, 2021-05-17:
What happens if lead car changes his mind and you'll accelerate into him?

pd0wm, 2021-05-17:
This stuff will get better once we switch to end-to-end longitudinal control. We won't be adding any hard-coded policy for this.

closed 2021-05-17 by pd0wm

adam-nelson, 2021-05-17:
Thanks. i had same opinion