commaai/openpilot #20927: ModuleNotFoundError: No module named 'Crypto'

LIHaoran-good, 2021-05-17:
I refer to this guideline that [Build openpilot for webcam](https://github.com/commaai/openpilot/blob/master/tools/webcam/README.md), when entering the compile instruction "USE_WEBCAM=1 scons -j$(nproc)", report ModuleNotFoundError: No module named 'Crypto',  details below:
(base) li@li-HD632-H81C:~/openpilot$ USE_WEBCAM=1 scons -j$(nproc)
scons: Reading SConscript files ...
ModuleNotFoundError: No module named 'Crypto':
  File "/home/li/openpilot/SConstruct", line 392:
    SConscript(['panda/board/SConscript'])
  File "/usr/local/lib/python3.8/dist-packages/SCons/Script/SConscript.py", line 654:
    return method(*args, **kw)
  File "/usr/local/lib/python3.8/dist-packages/SCons/Script/SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/usr/local/lib/python3.8/dist-packages/SCons/Script/SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/li/openpilot/panda/board/SConscript", line 144:
    certs = [get_key_header(n) for n in ["debug", "release"]]
  File "/home/li/openpilot/panda/board/SConscript", line 144:
    certs = [get_key_header(n) for n in ["debug", "release"]]
  File "/home/li/openpilot/panda/board/SConscript", line 63:
    from Crypto.PublicKey import RSA

**Expected behavior**
Openpilot could be built on PC successfully

Operating system: [Ubuntu 20.04]

pd0wm, 2021-05-17:
Did you correctly install the Pipfile? Looks like `pycryptodome` is missing

LIHaoran-good, 2021-05-17:
Hi@pd0wn, Thanks for your reply.
I reinstall the pycryptodome,  but entering the compile instruction "USE_WEBCAM=1 scons -j$(nproc), report another issue: 
scons: *** No SConstruct file found.
File"/usr/local/lib/python3.8/dist-packages/SCons/Script/Main.py", line 934, in_main

I don't know how to solve it.

pd0wm, 2021-05-17:
Looking at the paths from you report it doesn't look like you're using pyenv. Did you follow the instructions and ran the https://github.com/commaai/openpilot/blob/master/tools/ubuntu_setup.sh script?

You can also run everything from a virtualenv using pipenv. Use `pipenv install` followed by `pipenv shell` to active the environment.

The webcam build is tested in CI on every commit, so it must be something in your environment.

LIHaoran-good, 2021-05-17:
Thanks for your reply. It's very kind of you. 
You can see the details I report below. Maybe I should  follow the instructions and ran the ubtuntu_setup.sh script. Thanks very much.
(base) li@li-HD632-H81C:~/openpilot$ USE_WEBCAM=1 scons -j$(nproc)
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
mkdir -p cereal/gen/c/include && touch cereal/gen/c/include/c++.capnp.h cereal/gen/c/include/java.capnp.h
capnpc --src-prefix=cereal cereal/log.capnp cereal/car.capnp cereal/legacy.capnp -o c++:cereal/gen/cpp/
sh: 1: capnpc: not found
python3 cereal/services.py > cereal/services.h
clang++ -o cereal/messaging/msgq.os -c -std=c++1z -g -fPIC -O2 -Wunused -Werror -Wno-unknown-warning-option -Wno-deprecated-register -Wno-register -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-reorder-init-list -fPIC -I. -Iphonelibs/catch2/include -Iphonelibs/bzip2 -Iphonelibs/libyuv/include -Iphonelibs/openmax/include -Iphonelibs/json11 -Iphonelibs/curl/include -Iphonelibs/libgralloc/include -Iphonelibs/android_frameworks_native/include -Iphonelibs/android_hardware_libhardware/include -Iphonelibs/android_system_core/include -Iphonelibs/linux/include -Iphonelibs/snpe/include -Iphonelibs/nanovg -Iphonelibs/qrcode -Iphonelibs -Icereal -Icereal/messaging -Icereal/visionipc -Iopendbc/can -Iphonelibs/json11 cereal/messaging/msgq.cc
sh: 1: clang++: not found
clang++ -o cereal/visionipc/ipc.os -c -std=c++1z -g -fPIC -O2 -Wunused -Werror -Wno-unknown-warning-option -Wno-deprecated-register -Wno-register -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-reorder-init-list -fPIC -I. -Iphonelibs/catch2/include -Iphonelibs/bzip2 -Iphonelibs/libyuv/include -Iphonelibs/openmax/include -Iphonelibs/json11 -Iphonelibs/curl/include -Iphonelibs/libgralloc/include -Iphonelibs/android_frameworks_native/include -Iphonelibs/android_hardware_libhardware/include -Iphonelibs/android_system_core/include -Iphonelibs/linux/include -Iphonelibs/snpe/include -Iphonelibs/nanovg -Iphonelibs/qrcode -Iphonelibs -Icereal -Icereal/messaging -Icereal/visionipc -Iopendbc/can -Iphonelibs/json11 cereal/visionipc/ipc.cc
sh: 1: clang++: not found
clang++ -o cereal/visionipc/visionbuf.os -c -std=c++1z -g -fPIC -O2 -Wunused -Werror -Wno-unknown-warning-option -Wno-deprecated-register -Wno-register -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-reorder-init-list -fPIC -I. -Iphonelibs/catch2/include -Iphonelibs/bzip2 -Iphonelibs/libyuv/include -Iphonelibs/openmax/include -Iphonelibs/json11 -Iphonelibs/curl/include -Iphonelibs/libgralloc/include -Iphonelibs/android_frameworks_native/include -Iphonelibs/android_hardware_libhardware/include -Iphonelibs/android_system_core/include -Iphonelibs/linux/include -Iphonelibs/snpe/include -Iphonelibs/nanovg -Iphonelibs/qrcode -Iphonelibs -Icereal -Icereal/messaging -Icereal/visionipc -Iopendbc/can -Iphonelibs/json11 cereal/visionipc/visionbuf.cc
sh: 1: clang++: not found
clang++ -o cereal/visionipc/visionbuf_cl.os -c -std=c++1z -g -fPIC -O2 -Wunused -Werror -Wno-unknown-warning-option -Wno-deprecated-register -Wno-register -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-reorder-init-list -fPIC -I. -Iphonelibs/catch2/include -Iphonelibs/bzip2 -Iphonelibs/libyuv/include -Iphonelibs/openmax/include -Iphonelibs/json11 -Iphonelibs/curl/include -Iphonelibs/libgralloc/include -Iphonelibs/android_frameworks_native/include -Iphonelibs/android_hardware_libhardware/include -Iphonelibs/android_system_core/include -Iphonelibs/linux/include -Iphonelibs/snpe/include -Iphonelibs/nanovg -Iphonelibs/qrcode -Iphonelibs -Icereal -Icereal/messaging -Icereal/visionipc -Iopendbc/can -Iphonelibs/json11 cereal/visionipc/visionbuf_cl.cc
sh: 1: clang++: not found
cythonize cereal/messaging/messaging_pyx.pyx
scons: *** [cereal/gen/cpp/log.capnp.c++] Error 127
scons: *** [cereal/messaging/msgq.os] Error 127
scons: *** [cereal/visionipc/ipc.os] Error 127
scons: *** [cereal/visionipc/visionbuf.os] Error 127
scons: *** [cereal/visionipc/visionbuf_cl.os] Error 127
Compiling /home/li/openpilot/cereal/messaging/messaging_pyx.pyx because it changed.
[1/1] Cythonizing /home/li/openpilot/cereal/messaging/messaging_pyx.pyx
scons: building terminated because of errors.

pd0wm, 2021-05-17:
Closing this issue now since it's not really an issue with openpilot. If you need more assistance with setting up your environment please join the #dev-openpilot on our discord (discord.comma.ai)

closed 2021-05-17 by pd0wm