commaai/openpilot #20943: If the triangle label is not displayed?

johnlu6518, 2021-05-18:
Triangular labels sometimes appear, sometimes not. It seems to be related to the startup sequence of comma 2,
 If there is no triangle label, will it affect driving safety?

Device: Comma 2
Dongle ID: ec30903bb15db3ad

[image]

pd0wm, 2021-05-18:
The triangle indicates the detection of a lead car. Do you have any issues with lead car detection?

johnlu6518, 2021-05-18:
Because sometimes the triangle label does not appear, I must turn off the car first and start the engine again.(It seems to have to wait for the comma 2 to be completely turned on.）

johnlu6518, 2021-05-18:
If I turn on (Comma 2) and start the car engine at the same time, the triangle label will not be displayed normally.

pd0wm, 2021-05-18:
Do you have a specific route where the triangle does not appear?

adeebshihadeh, 2021-05-18:
Your car doesn't have longitudinal control, so the triangle should never show up. This is likely an issue with DSU detection.

johnlu6518, 2021-05-19:
For more information:

Device: Comma 2
Dongle ID: ec30903bb15db3ad
Route: ec30903bb15db3ad|2021-05-19--03-10-17

The strange finding  as follows:
You can't see the triangle on the device, but you can see the triangle on the laptop.

[image]
[image]
[image]

johnlu6518, 2021-05-19:
In addition, sometimes triangles can see on both the device and the laptop.

Route: ec30903bb15db3ad|2021-05-17--07-52-26
[image]
[image]

adeebshihadeh, 2021-05-19:
The discrepancy between explorer and the openpilot UI is expected. We recently modified the UI to only draw the lead triangle when openpilot controls longitudinal, but the same change wasn't made to explorer. @jwooning can we update explorer?

The real issue is showing the triangle in the openpilot UI. That route has openpilotLongitudinalControl=True, so this isn't a UI bug.

adeebshihadeh, 2021-05-19:
I see you have local modifications that add the smartDsu to the fingerprint. The smartDsu is causing your issue. Is there a reason you have it installed? It won't work on your car as-is.

johnlu6518, 2021-05-19:
Thank you for your reply, I don't remember any changes to smartDSU, can you take a screenshot or tell me where is different?

adeebshihadeh, 2021-05-19:
The diff from your device implies that you installed a smartDsu. The change itself doesn't cause the issue, it's the hardware.
```
diff --git a/selfdrive/car/toyota/values.py b/selfdrive/car/toyota/values.py\nindex 9c857f9..ddd5e24 100644\n--- a/selfdrive/car/toyota/values.py\n+++ b/selfdrive/car/toyota/values.py\n@@ -266,7 +266,7 @@ FINGERPRINTS = {\n   CAR.LEXUS_IS: [\n   # IS300 2018\n   {\n-    36: 8, 37: 8, 114: 5, 119: 6, 120: 4, 170: 8, 180: 8, 186: 4, 238: 4, 400: 6, 426: 6, 452: 8, 464: 8, 466: 8, 467: 5, 544: 4, 550: 8, 552: 4, 608: 8, 610: 5, 643: 7, 705: 8, 740: 5, 767: 4, 800: 8, 836: 8, 845: 5, 849: 4, 869: 7, 870: 7, 871: 2, 896: 8, 897: 8, 900: 6, 902: 6, 905: 8, 911: 8, 913: 8, 916: 3, 918: 7, 921: 8, 933: 8, 944: 8, 945: 8, 951: 8, 955: 8, 956: 8, 979: 2, 992: 8, 998: 5, 999: 7, 1000: 8, 1001: 8, 1005: 2, 1008: 2, 1009: 8, 1014: 8, 1017: 8, 1020: 8, 1041: 8, 1042: 8, 1043: 8, 1044: 8, 1056: 8, 1059: 1, 1112: 8, 1114: 8, 1161: 8, 1162: 8, 1163: 8, 1164: 8, 1165: 8, 1166: 8, 1167: 8, 1168: 1, 1176: 8, 1177: 8, 1178: 8, 1179: 8, 1180: 8, 1181: 8, 1182: 8, 1183: 8, 1184: 8, 1185: 8, 1186: 8, 1187: 8, 1189: 8, 1190: 8, 1191: 8, 1192: 8, 1196: 8, 1197: 8, 1198: 8, 1199: 8, 1206: 8, 1208: 8, 1212: 8, 1227: 8, 1235: 8, 1237: 8, 1279: 8, 1408: 8, 1409: 8, 1410: 8, 1552: 8, 1553: 8, 1554: 8, 1555: 8, 1556: 8, 1557: 8, 1561: 8, 1568: 8, 1569: 8, 1570: 8, 1571: 8, 1572: 8, 1575: 8, 1584: 8, 1589: 8, 1590: 8, 1592: 8, 1593: 8, 1595: 8, 1599: 8, 1648: 8, 1666: 8, 1667: 8, 1728: 8, 1745: 8, 1779: 8, 1904: 8, 1912: 8, 1990: 8, 1998: 8\n+    36: 8, 37: 8, 114: 5, 119: 6, 120: 4, 170: 8, 180: 8, 186: 4, 355: 5, 400: 6, 426: 6, 452: 8, 464: 8, 466: 8, 467: 5, 544: 4, 550: 8, 552: 4, 608: 8, 610: 5, 643: 7, 705: 8, 740: 5, 800: 8, 836: 8, 845: 5, 849: 4, 869: 7, 870: 7, 871: 2, 896: 8, 897: 8, 900: 6, 902: 6, 905: 8, 911: 8, 913: 8, 916: 3, 918: 7, 921: 8, 922: 8, 933: 8, 944: 8, 945: 8, 951: 8, 955: 8, 956: 8, 979: 2, 992: 8, 998: 5, 999: 7, 1000: 8, 1001: 8, 1008: 2, 1009: 8, 1014: 8, 1017: 8, 1020: 8, 1041: 8, 1042: 8, 1043: 8, 1044: 8, 1056: 8, 1059: 1, 1112: 8, 1114: 8, 1161: 8, 1162: 8, 1163: 8, 1164: 8, 1165: 8, 1166: 8, 1167: 8, 1168: 1, 1176: 8, 1177: 8, 1178: 8, 1179: 8, 1180: 8, 1181: 8, 1182: 8, 1183: 8, 1184: 8, 1185: 8, 1186: 8, 1187: 8, 1189: 8, 1190: 8, 1191: 8, 1192: 8, 1196: 8, 1197: 8, 1198: 8, 1199: 8, 1206: 8, 1207: 8, 1208: 8, 1212: 8, 1227: 8, 1235: 8, 1237: 8, 1279: 8, 1408: 8, 1409: 8, 1410: 8, 1552: 8, 1553: 8, 1554: 8, 1555: 8, 1556: 8, 1557: 8, 1561: 8, 1568: 8, 1569: 8, 1570: 8, 1571: 8, 1572: 8, 1575: 8, 1584: 8, 1589: 8, 1592: 8, 1593: 8, 1595: 8, 1599: 8, 1728: 8, 1745: 8, 1779: 8, 1904: 8, 1912: 8, 1990: 8, 1998: 8\n   },\n   # IS300H 2017\n   {\n@@ -867,6 +867,7 @@ FW_VERSIONS = {\n       b\'\\x018966353R1100\\x00\\x00\\x00\\x00\',\n       b\'\\x018966353R7100\\x00\\x00\\x00\\x00\',\n       b\'\\x018966353R8100\\x00\\x00\\x00\\x00\',\n+      b\'\\x018966353Q4000\\x00\\x00\\x00\\x00\',\n     ],\n     (Ecu.engine, 0x7e0, None): [\n       b\'\\x02353P7000\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00530J5000\\x00\\x00\\x00\\x00\\x00\\x00\\x00\\x00\',\n" ),
```

johnlu6518, 2021-05-19:
For this file, I modified two places as shown below: In addition, I did not install smartDSU. Can Lexus IS 200T install smartDSU?  I want to install it too if I can.
[image]
[image]

adeebshihadeh, 2021-05-19:
The explanations here are:
1. smartDsu is installed
2. something else uses the smartDsu address
3. 0x283 wasn't in the live fingerprint (unlikely because the DSU responded to the FW query)

If case 3 happened, it would be fixed in 0.8.4. I can't look into cases 1 and 2 since the bad route you posted doesn't have the full logs uploaded. I checked another route for case 2 and nothing is using that address. The most likely explanation is that a smartDsu is installed.

wiDavid, 2021-05-19:
I noticed something similar on my device. I am very new to open pilot (less than 36 hours), but I noticed that sometimes the triangle appears and sometimes it doesn't. When it does appear the revs on my 2018 NX300H get super weird, they kind of bounce up and down a couple thousand RPMs.

pd0wm, 2021-05-19:
@wiDavid Can you maybe share a fully uploaded route of both cases?

johnlu6518, 2021-05-20:
> The explanations here are:
>
> 1. smartDsu is installed
> 2. something else uses the smartDsu address
> 3. 0x283 wasn't in the live fingerprint (unlikely because the DSU responded to the FW query)
>
> If case [...]

> 這裡的解釋是：
> 
> 1. 安裝了smartDsu
> 2. 其他使用smartDsu地址
> 3. 0x283不在實時指紋中（這不太可能是因為DSU響應了FW查詢）
> 
> 如果情況3發生了，它將固定在0.8.4中。我無法調查情況1和2，因為您發布的錯誤路徑沒有上傳完整的日誌。我檢查了情況2的另一條路線，但沒有使用該地址。最可能的解釋是已安裝smartDsu。

I installed Blitz PowerThro, this device takes up the channel? Please refer to the link below:
https://www.blitz.co.jp/products/throcon/powerthro.html

johnlu6518, 2021-05-20:
> I installed Blitz PowerThro, this device takes up the channel? Please refer to the link below:
> https://www.blitz.co.jp/products/throcon/powerthro.html

I just provide information for your reference, maybe others have the same situation.

adeebshihadeh, 2021-05-20:
@johnlu6518 your external device is likely causing your issue. Etiher uninstall the device or hardcode this line to false https://github.com/commaai/openpilot/blob/master/selfdrive/car/toyota/interface.py#L326.

@wiDavid 0.8.4 shipped to release today and included a change relevant to your issue. If you still hit the issue after updating to 0.8.4, please open a new issue and make sure to fill out the bug report template.

johnlu6518, 2021-05-21:
> @johnlu6518 your external device is likely causing your issue. Etiher uninstall the device or hardcode this line to false [...]

I will try it, thank you very much.