commaai/openpilot #20970: Car was following the lead car properly but failed to brake for a lead car

adnanlanewla, 2021-05-19:
**Describe the bug**
Openpilot was engaged in the car. It was following the lead car properly and then all of a sudden, when lead car started braking OP failed to brake for the lead car. The comma's AEB warning system went on during this instance and I manually braked.

This is the third issue related to OP failed to brake for the lead car. Issue 20830 and Issue 20517 they both are very similar to this. 

**How to reproduce or log data**
Can't reproduce this bug

**Expected behavior**
OP should have braked while following the lead car

When OP is engaged it should brake for the lead car in front

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two] (comma2)
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin] (d2ef5aa690e849d0)
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin] [d2ef5aa690e849d0|2021-05-18--22-22-19--6]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)] (7s-14s)
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)] (0.8.3)
 - Car make/model: [e.g. Toyota Prius 2016]  (2020 Toyota Rav4)

**Additional context**
Here's the youtube video of the segment
https://youtu.be/MqtrUMJkPxQ

HaraldSchafer, 2021-05-19:
[image]

HaraldSchafer, 2021-05-19:
OP was definitely braking, it saw the car correctly and estimated it's speed well. You can see In the plot where you disengage.

It seems the longitudinal policy is just not aggressive enough when approaching a stopped car like this. We're working on that policy at the moment. Hopefully we can improve this soon.

adnanlanewla, 2021-05-20:
That's awesome good to hear that it's being worked on. Do you mind leaving the bug open until it's fixed ?

HaraldSchafer, 2021-05-20:
Sure

HaraldSchafer, 2021-05-20:
https://github.com/commaai/openpilot/issues/20976

TKassis, 2021-05-25:
I had the same thing happen again today. OP did not slow down for a stopped car and had to brake. It detects the car (evident by the yellow triangle that I see on the Comma but does not seem to recognize that it is stopped). 

Device: Comma Two
Dongle ID: cd21d47a9d46bdf3
Route: cd21d47a9d46bdf3|2021-05-25--17-07-05--12
Timestamp: Tue, 25 May @ 17:19:45
Version: 0.8.4 (running in laneless mode this time)
Car make/model: 2021 Toyota Corolla Hybrid

TKassis, 2021-05-26:
@HaraldSchafer I don't know if this is a Toyota-specific bug, but it seems a lot of people are encountering it noticeably after the 0.8.4 update.

TKassis, 2021-05-27:
Same thing happened again today. 0.8.3 used to slow down much better in these scenarios :-(

Device: Comma Two
Dongle ID: cd21d47a9d46bdf3
Route: cd21d47a9d46bdf3|2021-05-27--16-43-41--15
Timestamp: Tue, 27 May @ 16:59:26
Version: 0.8.4
Car make/model: 2021 Toyota Corolla Hybrid

adeebshihadeh, 2022-06-03:
@HaraldSchafer can we close?

iamjackg, 2022-07-27:
Something very similar happened to me yesterday. See #25295

HaraldSchafer, 2022-08-06:
Linking all these issues together 
https://github.com/commaai/openpilot/issues/25295

closed 2022-08-06 by HaraldSchafer