commaai/openpilot #20991: Move to C++ unlogger

adeebshihadeh, 2021-05-21:
We have parallel implementations of unlogger right now, the original python version and the new C++ version. The new unlogger is

adeebshihadeh, 2021-05-21:
@deanlee interested in this one? feel free to refactor LogReader and FrameReader as necessary, right now they seem unnecessarily tied to the replay use case.

deanlee, 2021-05-23:
I'll split it into several PRs

adeebshihadeh, 2021-09-21:
@deanlee startup seems to be super slow when the D/E cams are also uploaded

deanlee, 2021-09-22:
#22308 should be able to solve this

pd0wm, 2021-09-24:
Video looks strange when replaying using qcams now. Both on master and on #22195
[image]

deanlee, 2021-09-24:
@pd0wm that's strange. I can play qcamera normally on #22195 (demo route)

https://user-images.githubusercontent.com/27770/134628979-5ea438c5-2b5f-414f-a4c3-078f45356390.mp4

deanlee, 2021-09-24:
I modified Segment::Segment() (#22195 ) on my laptop to test fall back to qcamera. the result does not seem to have any problems
```
  // fallback to qcamera
  static int i = 0;
  if (++i % 2 == 0) {
    road_cam_path_ = files_.road_cam.isEmpty() ? files_.qcamera : files_.road_cam;
  } else {
    road_cam_path_ = files_.qcamera.isEmpty() ? files_.road_cam : files_.qcamera;
  }
```

deanlee, 2021-09-24:
- [ ] seeking stuck while segments is loading.
- [ ] framereader has a delay of 300-500ms after seeking.

pd0wm, 2021-09-24:
> @pd0wm that's strange. I can play qcamera normally on #22195 (demo route)
> I modified Segment::Segment() (#22195 ) on my laptop to test fall back to qcamera. the result does not seem to have any problems
> 
> ```
>   // fallback to qcamera
>   static int i = 0;
>   if (++i % 2 == 0) {
>     road_cam_path_ = files_.road_cam.isEmpty() ? files_.qcamera : files_.road_cam;
>   } else {
>     road_cam_path_ = files_.qcamera.isEmpty() ? files_.road_cam : files_.qcamera;
>   }
> ```
> 
>  1.mp4

I was using a C3 route. Maybe that's different?

deanlee, 2021-09-24:
> Can we control d/e cam using a command line flag? Think you don't need them most of the time.

Did you mean to load the driverCam/WideRoadCam only when there are -d or -e flags?

pd0wm, 2021-09-24:
Yeah, that sounds good.

pd0wm, 2021-09-28:
Noticed two more bugs while testing today:
 - Replay got stuck on a segment boundary, but that segment was already downloaded earlier according to the logs
 - When seeking into the route it started playing at that point, but no longer played in realtime but went as fast as it could.

deanlee, 2021-09-28:
> Qcameras seem to be broken on C3 footage only.

#22238 May be able to solve this problem

> Do you need a route for testing?

yes, that would be much better to have a c3 demo route

pd0wm, 2021-09-28:
Both problems might be related to a specific route I was replaying. Will do some more digging and let you know (`a8e8bf6a3864361b|2021-01-04--03-01-18`, t=840`)

pd0wm, 2021-09-28:
@deanlee I don't really understand what's going on. The route should be public so you can look as well. To reproduce seek to t=840. If you have the same problem as me it will play fast for a couple seconds, and then play normally. Then seek back to t=840 again and it will play faster for the whole segment. Any ideas?

deanlee, 2021-09-28:
> @deanlee I don't really understand what's going on. The route should be public so you can look as well. To reproduce seek to t=840. If you have the same problem as me it will play fast for a couple [...]

@pd0wm  it looks like an old bug related to keep time. if the us_begin<1ms, the replay will not sleep to wait. #22348  resolved this bug

pd0wm, 2021-09-28:
Can we make things more robust somehow? I don't think the code should be relying on the sleeps being ns accurate.

deanlee, 2021-09-28:
This case is a bug( the replay will not sleep  if us_begin>1), not the accurate. I think #22195  is enough

pd0wm, 2021-09-30:
Can we fix the visual glitches when seeking? I assume this is due to skipping past a keyframe?

deanlee, 2021-09-30:
yes it is. I'm going to fix it in #22217

deanlee, 2021-09-30:
The new demo route is not public 
```
"Error transferring https://api.commadotai.com/v1/route/4cf7a6ad03080c90%7C2021-09-29--13-46-36/files - server replied: FORBIDDEN"
failed load route "4cf7a6ad03080c90|2021-09-29--13-46-36" from server
```

pd0wm, 2021-09-30:
Should be public now!

deanlee, 2021-09-30:
#22238 can fix the transform problem on C3.
before:
[image]
after:
[image]

pd0wm, 2021-09-30:
Yeah, I'm aware. In the meantime you can also run with `TICI=1 ./ui`.

closed 2021-10-04 by pd0wm