commaai/openpilot #21008: Use Webcam tool  to run Openpilot

LIHaoran-good, 2021-05-24:
**Describe the bug**
I use the webcam tool to run openpilot on PC. We refer to the [guideline](https://github.com/commaai/openpilot/tree/master/tools/webcam) and connected the PC to two web cameras and CAN card(**We use CAN card instead of Panda**).   When input "PASSIVE=0 NOSENSOR=1 USE_WEBCAM=1 ./manager.py",  report " Connecting to panda
starting python selfdrive.loggerd.uploader
deviceState: Reader was evicted, reconnecting
"Error transferring https://api.commadotai.com/v1/devices/UnregisteredDevice/owner - server replied: NOT FOUND"
"Error transferring https://api.commadotai.com/v1.1/devices/UnregisteredDevice/ - server replied: NOT FOUND"

Details  error reported below :

(base) li@li-HD632-H81C:~/openpilot/selfdrive/manager$ PASSIVE=0 NOSENSOR=1 USE_WEBCAM=1 ./manager.py
System time valid
generating your personal RSA key
Generating RSA private key, 2048 bit long modulus (2 primes)
......................+++++
..........................................................................................................................+++++
e is 65537 (0x010001)
writing RSA key
getting pilotauth
Unable to register device, got 403
starting process ui
preimporting selfdrive.locationd.calibrationd
preimporting selfdrive.controls.controlsd
Type conversion already registered from type QList<QPair<QByteArray,QByteArray> > to type QtMetaTypePrivate::QSequentialIterableImpl
preimporting selfdrive.loggerd.deleter
preimporting selfdrive.logmessaged
preimporting selfdrive.pandad
preimporting selfdrive.locationd.paramsd
preimporting selfdrive.controls.plannerd
preimporting selfdrive.controls.radard
preimporting selfdrive.thermald.thermald
preimporting selfdrive.loggerd.uploader
manager start
{'environ': environ({'PASSIVE': '0', 'USE_WEBCAM': '1', 'NOSENSOR': '1', 'SHELL': '/bin/bash', 'SESSION_MANAGER': 'local/li-HD632-H81C:@/tmp/.ICE-unix/1569,unix/li-HD632-H81C:/tmp/.ICE-unix/1569', 'QT_ACCESSIBILITY': '1', 'COLORTERM': 'truecolor', 'XDG_CONFIG_DIRS': '/etc/xdg/xdg-ubuntu:/etc/xdg', 'XDG_MENU_PREFIX': 'gnome-', 'GNOME_DESKTOP_SESSION_ID': 'this-is-deprecated', 'CONDA_EXE': '/home/li/anaconda3/bin/conda', '_CE_M': '', 'PKG_CONFIG_PATH': '/usr/local/lib/pkgconfig::/usr/local/lib/pkgconfig', 'LC_ADDRESS': 'zh_CN.UTF-8', 'GNOME_SHELL_SESSION_MODE': 'ubuntu', 'LC_NAME': 'zh_CN.UTF-8', 'SSH_AUTH_SOCK': '/run/user/1000/keyring/ssh', 'XMODIFIERS': '@im=ibus', 'DESKTOP_SESSION': 'ubuntu', 'LC_MONETARY': 'zh_CN.UTF-8', 'SSH_AGENT_PID': '1540', 'GTK_MODULES': 'gail:atk-bridge', 'DBUS_STARTER_BUS_TYPE': 'session', 'PWD': '/home/li/openpilot/selfdrive/manager', 'LOGNAME': 'li', 'XDG_SESSION_DESKTOP': 'ubuntu', 'XDG_SESSION_TYPE': 'x11', 'CONDA_PREFIX': '/home/li/anaconda3', 'GPG_AGENT_INFO': '/run/user/1000/gnupg/S.gpg-agent:0:1', 'XAUTHORITY': '/run/user/1000/gdm/Xauthority', 'WINDOWPATH': '2', 'HOME': '/home/li', 'USERNAME': 'li', 'IM_CONFIG_PHASE': '1', 'LC_PAPER': 'zh_CN.UTF-8', 'LANG': 'en_US.UTF-8', 'LS_COLORS': 'rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:', 'XDG_CURRENT_DESKTOP': 'ubuntu:GNOME', 'VTE_VERSION': '6003', 'CONDA_PROMPT_MODIFIER': '(base) ', 'GNOME_TERMINAL_SCREEN': '/org/gnome/Terminal/screen/f1d54aa4_5e2d_4cec_bef4_5abff7816fad', 'INVOCATION_ID': 'e97091b69c6a4a1491f3cec1e4d6a8eb', 'MANAGERPID': '1370', 'LESSCLOSE': '/usr/bin/lesspipe %s %s', 'XDG_SESSION_CLASS': 'user', 'PYTHONPATH': '/home/li/openpilot', 'TERM': 'xterm-256color', 'LC_IDENTIFICATION': 'zh_CN.UTF-8', '_CE_CONDA': '', 'LESSOPEN': '| /usr/bin/lesspipe %s', 'USER': 'li', 'GNOME_TERMINAL_SERVICE': ':1.136', 'CONDA_SHLVL': '1', 'DISPLAY': ':1', 'SHLVL': '1', 'LC_TELEPHONE': 'zh_CN.UTF-8', 'QT_IM_MODULE': 'ibus', 'LC_MEASUREMENT': 'zh_CN.UTF-8', 'DBUS_STARTER_ADDRESS': 'unix:path=/run/user/1000/bus,guid=8652915b4cdf3b6bd122c95560ab1153', 'CONDA_PYTHON_EXE': '/home/li/anaconda3/bin/python', 'LD_LIBRARY_PATH': '/usr/local/cuda/lib64::/usr/local/lib', 'XDG_RUNTIME_DIR': '/run/user/1000', 'CONDA_DEFAULT_ENV': 'base', 'LC_TIME': 'zh_CN.UTF-8', 'JOURNAL_STREAM': '8:45152', 'OPENPILOT_ENV': '1', 'XDG_DATA_DIRS': '/usr/share/ubuntu:/usr/local/share/:/usr/share/:/var/lib/snapd/desktop', 'PATH': '/home/li/.pyenv/bin:/usr/local/cuda/bin:/home/li/anaconda3/bin:/home/li/anaconda3/condabin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin', 'GDMSESSION': 'ubuntu', 'DBUS_SESSION_BUS_ADDRESS': 'unix:path=/run/user/1000/bus,guid=8652915b4cdf3b6bd122c95560ab1153', 'LC_NUMERIC': 'zh_CN.UTF-8', '_': './manager.py', 'DONGLE_ID': 'UnregisteredDevice'})}
selfdrive/loggerd/bootlog.cc: bootlog to /home/li/.comma/media/0/realdata/boot/2021-05-24--13-27-31.bz2
starting daemon manage_athenad
starting python selfdrive.loggerd.deleter
starting python selfdrive.logmessaged
starting python selfdrive.pandad
starting python selfdrive.thermald.thermald
Connecting to panda
starting python selfdrive.loggerd.uploader
deviceState: Reader was evicted, reconnecting
"Error transferring https://api.commadotai.com/v1/devices/UnregisteredDevice/owner - server replied: NOT FOUND"
"Error transferring https://api.commadotai.com/v1.1/devices/UnregisteredDevice/ - server replied: NOT FOUND"
ui deleter logmessaged pandad thermald uploader
managerState: Reader was evicted, reconnecting
ui deleter logmessaged pandad thermald uploader
{"event": "STATUS_PACKET", "count": 0, "pandaState": null, "location": null, "deviceState": {"deviceState": {"freeSpacePercent": 67.44171905517578, "batteryPercent": 100, "batteryStatus": "Charging", "fanSpeedPercentDesired": 0, "started": false, "usbOnline": false, "startedMonoTime": 0, "thermalStatus": "green", "batteryCurrent": 0, "batteryVoltage": 0, "chargingError": false, "chargingDisabled": false, "memoryUsagePercent": 36, "cpuUsagePercent": 16, "networkType": "wifi", "offroadPowerUsageUwh": 0, "networkStrength": "unknown", "carBatteryCapacityUwh": 3000000, "cpuTempC": [0.0], "gpuTempC": [0.0], "memoryTempC": 0.0, "batteryTempC": 0.0, "ambientTempC": 0.0}, "logMonoTime": 10332900231293, "valid": true}}
"Error transferring https://api.commadotai.com/v1.1/devices/UnregisteredDevice/stats - server replied: NOT FOUND"
ui deleter logmessaged pandad thermald uploader
ui deleter logmessaged pandad thermald uploader
ui deleter logmessaged pandad thermald uploader
ui deleter logmessaged pandad thermald uploader
"Error transferring https://api.commadotai.com/v1.1/devices/UnregisteredDevice/ - server replied: NOT FOUND"
ui deleter logmessaged pandad thermald uploader
upload ('boot/2021-05-24--13-27-31.bz2', '/home/li/.comma/media/0/realdata/boot/2021-05-24--13-27-31.bz2') over wifi
{"event": "upload", "key": "boot/2021-05-24--13-27-31.bz2", "fn": "/home/li/.comma/media/0/realdata/boot/2021-05-24--13-27-31.bz2", "sz": 1079}
checking 'boot/2021-05-24--13-27-31.bz2' with size 1079
uploading '/home/li/.comma/media/0/realdata/boot/2021-05-24--13-27-31.bz2'

**Expected behavior**

UI should show the road webcam's view  successfully

**Additional context**

Operating system: [Ubuntu 20.04]

adeebshihadeh, 2021-05-24:
openpilot goes onroad on the rising edge of ignition, which comes from the panda. If you just want the camera view or to use other hardware besides panda, you'll need to modify the code.

closed 2021-05-24 by adeebshihadeh

LIHaoran-good, 2021-05-24:
Thanks for your reply.
I also have supporting vehicle now, and I want to replace Panda with CAN Card,  which specific code module do I need to modify, looking forward to your reply.