commaai/openpilot #21033: My vehicle no longer works after upgrading

Verylukyguy, 2021-05-26:
**I upgraded to 0.8.4 and my Comma Two no longer works with my vehicle.**

I received my Comma Two in Aug 2020 and my 2018 GMC Acadia's fingerprint was slightly different from the already "Community Supported" Acadia that was added by CryptoKylan in version 0.5.7

I contacted Comma.ai and Virtually Chris created a new branch for me with my vehicle's fingerprint on Oct 14, 2020 (GMC-Acadia-fp) and when I tried to use that branch, I was getting constant "Steering Temporarily Unavailable" errors when trying to engage OpenPilot or while driving with OpenPilot.

The Community pointed my to JShuler because he had solved the same bug for the Chevy Bolt port.

Jason was very helpful and created a branch for me on Oct 21, 2020 and I have been using it ever since.  https://github.com/JasonJShuler/openpilot/tree/devel-luckyguy

I was hoping that the fork/branch that I had been using for many month would have been upstreamed, but Jason Shuler was banned.

My Comma Two's Dongle ID is:  e38a02cae1ef1e0c

My vehicle is a 2018 GMC Acadia Denali w/ FE5 Suspension

I have uploaded a few drives for you to review.

I'm not a computer programmer, but I'm not totally illiterate. 

It seems like if someone would upstream the fork/branch that I've been using, maybe other compatible GM vehicle might benefit.

I copied it to my GitHub at:  https://github.com/Verylukyguy/openpilot

Thank you for helping me.

I have attached my fingerprint, if it helps.

190: 6, 192: 5, 193: 8, 197: 8, 199: 4, 201: 6, 208: 8, 209: 7, 211: 2, 241: 6, 249: 8, 288: 5, 289: 1, 290: 1, 298: 8, 304: 8, 309: 8, 313: 8, 320: 8, 322: 7, 328: 1, 352: 7, 368: 8, 381: 8, 384: 8, 386: 8, 388: 8, 393: 8, 398: 8, 413: 8, 417: 7, 419: 1, 422: 4, 426: 7, 431: 8, 442: 8, 451: 8, 452: 8, 453: 6, 454: 8, 455: 7, 458: 8, 460: 4, 462: 4, 463: 3, 479: 3, 481: 7, 485: 8, 489: 5, 497: 8, 499: 3, 500: 6, 501: 8, 508: 8, 510: 8, 512: 3, 530: 8, 532: 6, 534: 2, 554: 3, 560: 8, 562: 8, 563: 5, 564: 5, 567: 5, 568: 2, 573: 1, 608: 8, 609: 6, 610: 6, 611: 6, 612: 8, 613: 8, 647: 6, 707: 8, 715: 8, 717: 5, 753: 5, 761: 7, 789: 5, 800: 6, 801: 8, 803: 8, 804: 3, 805: 8, 832: 8, 840: 5, 842: 5, 844: 8, 866: 4, 869: 4, 880: 6, 961: 8, 969: 8, 977: 8, 979: 8, 985: 5, 1001: 8, 1003: 5, 1005: 6, 1009: 8, 1017: 8, 1020: 8, 1033: 7, 1034: 7, 1105: 6, 1217: 8, 1221: 5, 1225: 8, 1233: 8, 1249: 8, 1257: 6, 1265: 8, 1267: 1, 1280: 4, 1296: 4, 1300: 8, 1322: 6, 1328: 4, 1417: 8, 1906: 7, 1907: 7, 1912: 7, 1914: 7, 1918: 7, 1919: 7, 1920: 7, 1930: 7

adeebshihadeh, 2021-05-27:
Closing as I wouldn't consider this a bug since your vehicle never worked on stock openpilot. Feel free to open a PR, and we'll help get this in an upstreamable state.

closed 2021-05-27 by adeebshihadeh

Verylukyguy, 2021-05-27:
I was instructed by Chris to open this issue and he said that Shane was going to help and try to do this.  I don't know how to do a pull request.

adeebshihadeh, 2021-05-27:
Chris misunderstood your issue. Vehicle support largely comes from the community, particularly for makes that are community supported.

Verylukyguy, 2021-05-27:
My vehicle worked on stock OpenPilot after my fingerprint was added, but I got Steering Temporarily Unavailable errors and JShuler's branch fix that.  It's hard to believe how little support is actually available.

Verylukyguy, 2021-05-27:
Shuler had me up and running in about a half hour.  I'll give you $100 if you can get me a working version of 0.8.4

ShaneSmiskol, 2021-05-28:
Can you share the branch that added support? I can't guarantee anything, but if I get any time I may be able to look into this if it's a simple fix.

ShaneSmiskol, 2021-05-28:
I recommend asking Shuler for the minimum required changes to get you up and running, and then you can `git cherry-pick` that minimal commit for each new version of openpilot. Cherry picking is essentially applying the changes (diff) of a commit onto a branch

Verylukyguy, 2021-05-28:
This is the branch that I have been using for 7 months and I have never had a "Steering Temporarily Unavailable" error.
I would appreciate any help, because I can't write or edit code.

https://github.com/JasonJShuler/openpilot/tree/devel-luckyguy

I have attempted to contact Jason a couple times and he has been devastated by his ban; he used to passionately engaged and willing to help anyone, but he hasn't responded.  He told me that the problem was when a lane keep assist message is sent less than 20 milliseconds after the one before it and it gets out of sync and his code tightens up the timing by moving the message generation onto the panda and using the timer on the panda.