commaai/openpilot #21038: ui dispaly NO PANDA when input "PASSIVE=0 NOSENSOR=1 USE_WEBCAM=1 ./manager.py"

LIHaoran-good, 2021-05-26:
**Describe the bug**
I use the webcam tool to run openpilot on PC. We refer to the guideline and connected the PC to two web cameras and CAN card(We use CAN card instead of Panda). When input "PASSIVE=0 NOSENSOR=1 USE_WEBCAM=1 ./manager.py", report " Connecting to panda
starting python selfdrive.loggerd.uploader
deviceState: Reader was evicted, reconnecting
"Error transferring https://api.commadotai.com/v1/devices/UnregisteredDevice/owner - server replied: NOT FOUND"
"Error transferring https://api.commadotai.com/v1.1/devices/UnregisteredDevice/ - server replied: NOT FOUND"
The ui interface is as follows and how to solve it, looking forward to your reply. 
[image]

**Expected behavior**
The UI could display Openpilot correctly

**Additional context**

Operating system: [e.g. Ubuntu 20.04]

pd0wm, 2021-05-26:
What is a CAN card? Openpilot only works using a panda.

LIHaoran-good, 2021-05-28:
CAN card is also used to communicate with vehicle, which is the same as Panda

LIHaoran-good, 2021-05-28:
@pd0wm Hi,  I see the **OpenCL Driver** is mentioned in [guideline](https://github.com/commaai/openpilot/tree/master/tools/webcam), so I want to ask what is the function of the **OpenCL Driver**?
Thanks for your reply.

adeebshihadeh, 2021-05-29:
If you want to use your "CAN card", you'll need to modify boardd. OpenCL is used in camerad and modeld.

closed 2021-05-29 by adeebshihadeh

LIHaoran-good, 2021-05-29:
@adeebshihadeh Hi， how to install OpenCL Driver, do you have the guideline, thanks.