commaai/openpilot #21054: Red Alert (Steering Temp. Unavilable) on Winding Turns when Manually Turning

jsegill, 2021-05-27:
Since 0.8.4, release-2 was installed on my C2, I am getting many red alerts stating "Take Control Immediately- Steering Temporarily Unavailable". Mostly getting the take control immediately red alerts when steering the car by hand in a winding corner, most at lower speed (17-35mph). Previous versions of OP (0.8.3 and below) would give me a yellow alert when engaged on the same winding turns. The old yellow alert would say "steering unavailable" or "steering torque maxed out" and would not go past that stage into a red alert. Whenever the yellow of red alert was happening, I was usually hand steering the vehicle. The new red alert in winding turns doesn't make much sense since I am hand steering and well have hands on the steering wheel already. Also it feels as though OP still has EPS control when these red alerts are occurring. I feel like the excessive amount of red alerts directly related to this "Steering Temporarily Unavailable" is devaluing the importance of getting a red alert. The previous yellow worked well for getting attention when maxing out the EPS torque. 

Ideally find a winding road that supports a speed of 35mph or less (lesser the speed the more the red alert occurs). With OP engaged, start hand steering the vehicle through the winding turn and on an apex of a turn give the steering wheel a slight over correction. This red alert happens when not over correcting in turns as well, but it's the easiest way to make the issue occur. Another way to make the issue repeat is to be in stop and go traffic on a winding off ramp and make steering adjustments at low speed (20mph or less). Hand turning the steering wheel too much will cause the same red alert. 

Yellow alert to get driver to put hands on steering wheel. Yellow stating something like "Steering Unavailable" (if that's accurate), and "Take Control".

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 9945372986050f92
 - Route: 9945372986050f92|2021-05-27--10-19-25
 - Timestamp: 10 minutes into drive
 - Version: 0.8.4 release-2
 - Car make/model: Toyota Corolla Hatchback 2019

**Additional context**

I have additional drives on my daily C2 from driving in the city and this is when I first noticed the issue. Those drives will have more realistic driving. The route segment provided in this report was me trying to provoke the issue on a separate C2. 

[image]

https://user-images.githubusercontent.com/75038839/119859734-a630c600-bee3-11eb-90d8-053d4fa1f934.mp4

srieg, 2021-05-28:
I also have been experiencing this same issue on 0.8.4. Happy to track down an example if helpful

adeebshihadeh, 2021-05-28:
Your route isn't fully uploaded. Can you connect your device to wifi and ensure the "Upload Raw Logs" toggle is enabled?

jsegill, 2021-05-28:
Just enabled Upload Raw Logs. The device is currently on.

adeebshihadeh, 2021-05-28:
This started in 0.8.4 because we changed the alert to be more aggressive and made a separate alert for unavailable steering when you're overriding. In your route, `steerWarning` occurs before the threshold is hit for `steeringPressed`.  We'll need ti fix up the alert priorities to fix this properly.

[image]

jsegill, 2021-05-28:
Would additional drives with this happening help?

mysliwiec-tech, 2021-06-12:
Same happens for me. I'll just provide a drive that I got it logged. I cannot identify what is causing the issue. Seems random to me.

Device/Version information (please complete the following information):

Device: comma two
Dongle ID: bd1aa84276ac73ae
Route: bd1aa84276ac73ae|2021-06-12--13-57-33
Timestamp: 14:05:29 (in the middle of the recording)
Version: 0.8.5 release-2 (but was happening since 0.8.4 release-2)
Car make/model: Toyota Corolla TS 2020 Hybrid

snowwolf3388, 2021-06-13:
I'm actually on kegman-083, but when I override steering for a high torque turn, I often see this message (in 0.8.3 it is not a red alert, but now I'm afraid to upgrade to 084/085 because it will scare the crap out of my family). Here's a route from today. Sorry it is not release2, but hopefully it is still useful. Would love to see this fixed.

Device: comma two
Dongle ID: 70b939e655d0e90c
Route: 70b939e655d0e90c|2021-06-12--13-51-27
Timestamp: 14:07:40 
Version: kegman-083
Car make/model: 2020 Honda Odyssey EX-L

jcwillia1, 2021-06-18:
Same exact issue. Same car. Sonata turning left from Masters to 96th in castleton / Indianapolis indiana will trigger the alarm. Turning right from 96th to Hazel Dell also triggers it.

never had problems before. 

Interestingly there’s a pretty good curve on Hazel Dell approaching 96th from the north at 40mph that does not trigger the alarm in either direction.

adeebshihadeh, 2021-07-10:
Fixed in #21541.

closed 2021-07-10 by adeebshihadeh