commaai/openpilot #21055: Do same checks in panda and carstate for Bosch and CRV Hybrid

ShaneSmiskol, 2021-05-27:
There's about 50 mismatches on the Honda CRV Hybrid for `brakePressed` between carState and panda

pd0wm, 2021-05-28:
Are you using a route from before https://github.com/commaai/openpilot/commit/4215f63769fd6f350194cf5b6c9607cce40e18e0 was merged? It was using the wrong signal in the panda for a while.

adeebshihadeh, 2021-05-29:
This test just replays the CAN messages through openpilot and panda, so that wouldn't affect this.

loveloveses, 2021-06-05:
I think the mismatches is as the ibootser use for brake when use acc
the ibooster can give -0.2g for brake，when the car need more for brake，the VSA will simulation an brake，and the brakelight will light on

loveloveses, 2021-06-05:
i have open an issue as #19838，but it is closed

adeebshihadeh, 2022-01-27:
This is covered by #23643

closed 2022-01-27 by adeebshihadeh