commaai/openpilot #21057: Intermittant "No Panda" issue requireing reboot of C2

nabeelr, 2021-05-27:
**Describe the bug**

Occasionally, my C2 will say "No Panda" along with a CAN error message. The car (a 2019 VW Golf R) will also throw a bunch of errors at that time with a bunch of modules. A reboot is needed to correct the issue, on the C2. Sometimes a restart of the car is needed to clear it's issues too. The C2's panda will remain offline without rebooting the C2, despite what you do to the car.

**How to reproduce or log data**

Unable to reliably reproduce. Driving data exists however.
Issue occurred at the end of the drive on this segment 903210ecc0cb706b|2021-05-25--21-13-22--80
OP was engaged right up until the end of the drive, despite Explorer saying it was disengaged.

**Expected behavior**

For Panda not to go offline randomly.

**Device/Version information (please complete the following information):**
 - Device: Comma2
 - Dongle ID: 903210ecc0cb706b
 - Route: 903210ecc0cb706b|2021-05-25--21-13-22--80
 - Timestamp: 22:34, right at the end of the drive. After the footage ended, about 10 seconds later in that roundabout, OP disengaged, showed the red "CAN Error" message and disengage warning, the car's LKA and ACC went offline, and OP screamed until I rebooted it about 30 seconds later.
 - Version: [commit 0.8.4
 - Car make/model: 2019 Golf R Manual

**Additional context**

Lots of errors logged in the cars ABS, LKA, ECU, EPS, and BCM modules after this this issue. These errors do not occur without the C2 present.

vanillagorillaa, 2021-05-28:
Do you have a SIM card in your device? Does this seem to happen in the same places when driving around?

nabeelr, 2021-05-28:
I do have a SIM card in it, but it doesn't seem to happen in the same place ever. It's only happened a couple times so far.

When I first got my C2, I did have an issue with tower changes causing issues, but that has since been fixed. I used to be able to reproduce it by driving on a specific road, however driving that road doesn't cause this issue.

My C2 DID hard lock a couple times in the winter, but after re-flashing NEOS it seems to have stopped doing that. This is fairly new, only in the last couple weeks has it done it (Maybe since 0.8.4? but it may have happened on 0.8.3 too. My memory is shit sooo...), and it's done it twice so far. The first time I ignored, and only took note the second time it happened. 

I saw this issue: #20872 but it didn't seem similar enough since it doesn't SEEM to be tied to the cell tower switchover like before... BUT they did link to this issue: #19874 which I had before and even commented on, but like I said, was fixed for me by re-flashing NEOS. Maybe it's the same issue and the "fix" was co-incidence, or this is a different expression of the same issue? IDK...

It seemed too different to be related, hence the separate ticket.

nabeelr, 2021-05-29:
it literally just happened again. I was in the hills near my house, we're cell phone reception is generally pretty spotty. Not sure if that's related, but it was a concern that someone had. I'm still in the area, so I'll see if it happens again on my way back.￼￼

I'll post the route information a little bit later today.￼

nabeelr, 2021-05-30:
So, as I mentioned earlier, it happened again today, BUT there were also two other issues that occurred after a reboot shortly after in the same area. Though these presented in a new way that I don't think I've seen happen before. Not sure if they are related or not, but I can go back and test again to see if they happen in that area again.

The first thing that went wrong was the issue I described earlier, CAN connection error, red warning screen, No Panda afterwards, though the panda firmware still showed up in the Developer tab.
The only way I was able to correct this issue was with a reboot of the C2.
This DID occur in an area with poor cell reception, and the C2 was constantly saying that it was gaining and losing cell service. Though the other times it has happened, weren't in areas of poor cell service, and never seemed to be reproducible before. Maybe it will be on this road? I can try again tomorrow.
This occurred at: 903210ecc0cb706b|2021-05-29--17-04-20--30 at approximately 17:35:01
[image]
[image]

Later that same drive (though after rebooting the C2), I saw it do the two other things I mentioned in: 903210ecc0cb706b|2021-05-29--17-46-50--5 at approximately 15:52:31 I got an error that I wasn't able to read, but I suspect it was another CAN Error. Panda stayed online, but later trying to re-engage showed an error saying something about controls mismatch somewhere after 17:52:56.

Approximately at 17:53:50, I turned the ignition off and right back on again (not giving the car enough time to reset or for modules to go to sleep) and was able to drive again with my C2 without incident for the rest of the time back home.

During the ride back home, after the reboot of the C2 I kept an eye on the "connected" status and cellular signal strength of the C2 and saw it go from red to yellow to white several times, without issue or incident. I also observed the C2 gain and lose cell service, switch back and forth between 2G, 3G and LTE without issue.

I'll probably do this drive again tomorrow, let me know if you want me to try it again, and I can make it a priority.

Hope this helps!

nabeelr, 2021-05-31:
Did a fair bit of testing. Does seem to be SIM card related, so this issue might be one and the same as #20872 
I'll post an update tomorrow morning.

nabeelr, 2021-05-31:
So, I managed to find a stretch of road where I could make the issue occur consistently and compared with and without the sim card. which definitely seemed to cause the issue. At one point, the C2 lost the panda entirely, and I had to disconnect power from the C2 entirely for it to come up again.

I did the road 3 times each, up, back down, and up again.

Here are the drives:
903210ecc0cb706b|2021-05-30--22-40-33--27
903210ecc0cb706b|2021-05-30--23-09-52--0
903210ecc0cb706b|2021-05-30--23-14-29--0

Without the SIM I was able to do all 3 drives together, then popped the sim back in for a 4th run, where it died :
903210ecc0cb706b|2021-05-30--23-19-40--0

The drive with the SIM card starts at approximately 23:32:50, and ends at the end of the clip.

The problem manifested in many different ways. From the panda going offline, to the C2 thinking the car was off, to getting a controls mismatch error, or a CAN communication error.

adeebshihadeh, 2021-06-01:
Thanks for looking into this and isolating it to the SIM card/cellular connection. Yes, let's close this and add the info to the existing issue. I'll look at your routes tomorrow. We're actively looking into this issue, and hope to ship a fix soon in 0.8.5.

nabeelr, 2021-06-01:
Cool, will do. I can very consistently reproduce the issue, so let me know if you want me to do any testing.

closed 2021-06-01 by nabeelr