commaai/openpilot #21062: "Unable to download updates, Ensure the software is correctly installed"

TKassis, 2021-05-28:
**Describe the bug**

As soon as I parked after a drive today I got a red alert message saying:

"Unable to download updates, Ensure the software is correctly installed"

As far as I know I'm running the latest stock OP, so not sure why this message is even coming up.

**How to reproduce or log data**

Drive, park.

**Expected behavior**

Not to display this message. Especially when there is no available update to begin with.

**Device/Version information (please complete the following information):**
 - Device: Comma Two
 - Dongle ID: cd21d47a9d46bdf
 - Route: cd21d47a9d46bdf3|2021-05-27--16-43-41--0
 - Timestamp: at the end after I turned off my car
 - Version: 0.8.4
 - Car make/model: 2021 Toyota Corolla Hybrid

**Additional context**

[image]

bakgyber, 2021-05-28:
> **Describe the bug**
>
> As soon as I parked after a drive today I got a red alert message saying:
>
> "Unable to download updates, Ensure the software is correctly installed"
>
> As far as I know I'm [...]

So is my Camry.  I thought it was only me.

nabeelr, 2021-05-28:
This happened to me too, from what I saw on the Comma discord, it's a known issue that's actively being fixed.

TKassis, 2021-05-28:
It seems they fixed it this morning and pushed a hotfix to everyone.

closed 2021-05-28 by TKassis