commaai/openpilot #21078: 'Steering Temporarily Unavailable' Master warning when speed below 32mph with disengage of stock ACC

sunilm1992, 2021-05-29:
**Describe the bug**

<When speed below 32mph on Kia Optima 2017 (without MDPS harness), comma device displays master warning (red flash with loud beep) - 'steering temporarily unavailable' and then after 2-3 seconds disables cruise control. This effectively kills stop and go cruise control - this is a new issue in 0.8.4 and onwards. Behaviour in 0.8.3 was that you wouldn't get OP steering below 32mph be cruise control would work as normal (stock ACC) and once speed rose above 32mph, steering from OP would reactivate - I found this to be acceptable.>

**How to reproduce or log data**

<Crossing above and below 32mph in OP >

**Expected behavior**

<For no master warning below 32mph - to work as previous 'steering limit reached, take control' message and for ACC not to be disabled as there is a risk of a collision if the user expects ACC to be on and it disengages because of this>

**Device/Version information (please complete the following information):**
 - Device: EON Gold
 - Dongle ID: c7f89112bca83879
 - Route: c7f89112bca83879|2021-05-29--13-06-46
 - Timestamp: 14:11
 - Version: 0.8.4 stock
 - Car make/model: Kia Optima 2017

**Additional context**

Thank you for reviewing my bug report. I think this is a step backwards in functionality and has the potential to cause an accident

xps-genesis, 2021-05-29:
Possibly doesn't have correct min_steer_speed. Possibly OP is requesting steering below 32mph and you car is setting LKAS fault. Please verify your car is getting recognized as the correct model in the fingerprints. and the min_steer_speed value is correct for your car.

sunilm1992, 2021-05-29:
> Possibly doesn't have correct min_steer_speed. Possibly OP is requesting steering below 32mph and you car is setting LKAS fault. Please verify your car is getting recognized as the correct model in the fingerprints. and the min_steer_speed value is correct for your car.

Sorry to be a noob, can you give me a pointer on how to check minimum steer value?

sunilm1992, 2021-05-29:
in car explorer / admin portal it is recognising as Kia Optima the same way as it did previously

ErichMoraga, 2021-05-29:
`chrsfr#1406` said he reverted a commit that was made to `selfdrive/car/hyundai/carstate.py`, so I incorporated that "fix" into my `084` branch, which @sunilm1992 confirmed worked...
https://github.com/ErichMoraga/openpilot/commit/e6b1096fbc85db2a18096b2ec2cdbf31f67d78cf
https://discord.com/channels/469524606043160576/524592892627517450/848327720563769375

pd0wm, 2021-06-02:
Looks like xps is right and `minSteerSpeed` is not set for the Optima. Can you check if adding `ret.minSteerSpeed = 32 * CV.MPH_TO_MS` to `interface.py` fixes it? Is this something that's true for all Optima trims/years?

This is a better fix than hiding the fault condition, since by setting `minSteerSpeed` it should also show an alert when openpilot is not steering.

ErichMoraga, 2021-06-02:
I've created this branch, which I'll have `@chrsfr#1406` test...
https://github.com/ErichMoraga/openpilot/tree/084_optima

sunilm1992, 2021-06-02:
> Looks like xps is right and `minSteerSpeed` is not set for the Optima. Can you check if adding `ret.minSteerSpeed = 32 * CV.MPH_TO_MS` to `interface.py` fixes it? Is this something that's true for [...]

It's only for 2017-2019
2019 onwards has steering down to 0mph

pd0wm, 2021-06-03:
Can we detect this difference based on EPS firmware version? We either need to maintain a list of 2017-2019 EPS firmware versions (like we do for some special rav4 EPS, https://github.com/commaai/openpilot/blob/master/selfdrive/car/toyota/interface.py#L216), or split it out as a completely separate car model.

ErichMoraga, 2021-06-04:
@sunilm1992 reported back on your suggested fix earlier (ret.minSteerSpeed = 32 * CV.MPH_TO_MS), and that didn't work.  He said...
"New fix goes back to previous state unfortunately... constant warning on screen with red warning below 32 and disengagement."

If we do something based on his EPS f/w version, @sunilm1992 will have to disclose his that info. to me, or give me access to his DongleID `c7f89112bca83879`.  Of course @pd0wm should already have access...

ShaneSmiskol, 2021-06-04:
Perhaps we should be killing steering torque requests when actually below the min steer speed? Right now it just shows an alert, but still sends commands to the car: https://github.com/commaai/openpilot/blob/master/selfdrive/car/hyundai/carcontroller.py#L53

ErichMoraga, 2021-06-04:
I've updated my `084_optima` branch accordingly, and advised @sunilm1992 to try it out.

sunilm1992, 2021-06-04:
so the latest change works but:

slight change to make if possible? - in the kia optima steering is unavailable below 32mph (as displayed by the car speedometer), in this latest update OP doesn't request steering when the OP speed (wheel speed sensor) is 32 or below, but in reality this should be 29 mph (as speedometer 32 = 28 OP and you want 1mph leeway)
[11:40 PM]
so is it possible to change the code to say not to request steering if OP speed is less than or equal to 28mph on wheel speed sensor?

gregjhogan, 2021-06-04:
> Perhaps we should be killing steering torque requests when actually below the min steer speed? Right now it just shows an alert, but still sends commands to the car: https://github.com/commaai/openpilot/blob/master/selfdrive/car/hyundai/carcontroller.py#L53

https://github.com/commaai/openpilot/pull/20657 was merged yesterday which prevents sending torque in this situation (but perhaps cutting torque after the fault occurs will still result in a red alert)

sunilm1992, 2021-06-12:
> #20657 was merged yesterday which prevents sending torque in this situation (but perhaps cutting torque after the fault occurs will still result in a red alert)

hi in 085 this is still an issue
when speed < 32mph the red alert comes on but ACC is not disengaged as before

would be really grateful if this could be fixed!! makes OP unusable

sunilm1992, 2021-06-12:
> `chrsfr#1406` said he reverted a commit that was made to `selfdrive/car/hyundai/carstate.py`, so I incorporated that "fix" into my `084` branch, which @sunilm1992 confirmed worked...
> [ErichMoraga@e6b1096](https://github.com/ErichMoraga/openpilot/commit/e6b1096fbc85db2a18096b2ec2cdbf31f67d78cf)
> https://discord.com/channels/469524606043160576/524592892627517450/848327720563769375

i'm going back to Erich's 084 branch which fixes the issue!

jyoung8607, 2021-12-15:
The user that opened this bug contacted me on Discord for help. They were confused over what the next step was, so just summarizing the findings already in this bug: openpilot needs to be taught the proper minSteerSpeed for the older Optimas.

* Before 0.8.4: openpilot steered down to zero, EPS faulted, openpilot didn't notice (which is bad), ACC remained engaged down to zero
* Starting with 0.8.4: openpilot steered down to zero, EPS faulted, openpilot noticed (which is better albeit inconvenient for this end user) and eventually disengaged ACC
* Desired situation: openpilot steers down to openpilot/EPS limit, stops steering below that and does not trigger an EPS fault, ACC remains engaged down to zero

Torque shouldn't applied below minSteerSpeed beginning with the non-Kia-Niro portion of #21481, so that much should be taken care of even if #20657 wasn't enough (as alluded to earlier in the bug history). The user no longer has their C2, but has a C3 on the way. When they have it, I can work with them to test the fix.

After that, it'll be a matter of identifying which Optimas it applies to. @ErichMoraga saw data above that indicates we might not get EPS firmware, but this should be rechecked after the recent fixes to HKG UDS queries. If the user's car still isn't returning EPS firmware, we could see if all minSteerSpeed=0 cars DO return firmware and infer it from that. Or, fwdCamera firmware might be a good proxy. Splitting CARs just to infer the model-year-group from powertrain and ABS firmware seems wrong to me.

adeebshihadeh, 2022-06-04:
Looks like the current optima platform covers two generations. @sshane let's split them up

sshane, 2022-06-10:
We need to fix the UDS queries for Hyundai first I believe, we only have one fw version for all ecus but two, and I checked and our Optima here doesn't return eps most of the time (if at all)

sshane, 2022-09-20:
Your car is now under the `KIA OPTIMA 2016` fingerprint. If you're able to try master, let us know how it goes!

closed 2022-09-20 by sshane