commaai/openpilot #21079: Alert says "connect to internet to update or won't engage" yet no prompt for update and still engages

alexklibisz, 2021-05-29:
**Describe the bug**

C2 tells me to connect to internet to check for updates or openpilot will not engage in 0 days. I connect to internet, make sure car is off, make sure wifi is good, wait like 20 minutes, reboot a few times, try another wifi network, still same alert and no update prompt.

**How to reproduce or log data**

**Expected behavior**

Comma should either get better at actually downloading the update, or, preferably, give me an explicit button to click to go fetch the update. It's a very annoying UX to sit there and wait and try random things with no feedback at all.

**Device/Version information (please complete the following information):**
 - Device: Comma two
 - Dongle ID: e7691e62498e6411
 - Route: N/A
 - Timestamp: N/A
 - Version: 0.8.2
 - Car make/model: Corolla 2017

**Additional context**

In general this seems like a very buggy part of the system. A related issue, I've had the "will not engage after 0 days" alert for basically every version for over a year, yet Comma definitely does still engage.

alexklibisz, 2021-05-29:
A full re-install ended up getting me onto 0.8.4. That's technically not an _update_ in the true sense of how it should work, but good enough for now. Would be great to get some more insight into what's going wrong here.

pd0wm, 2021-05-31:
- 0.8.5 will include UI improvements for the updater. This let's you see the last update time and manually initiate an update.
 - Even with the maximum time between updates exceeded openpilot still lets you engage on the first drive after boot. This is for situations where your device has been off for a while (e.g. airport parking lot) so you can still drive until it had the chance to actually look for an update.
 - Sounds like your openpilot folder got corrupted. Some of our version check functions still write to the `.git` folder, which can cause corruption if your device loses power at the wrong time. We're working on getting rid of the last few things that touch the disk.

alexklibisz, 2021-05-31:
> Another issue we have seen with other users is that git might be blocked on (one of) your wifi networks the device is connected to. Any exception when trying to update causes the offroad alert to show. If you run a developer branch (e.g. master-ci) you get more details about the specific exception in the alert. If you keep having issues it might be worth it to try that, it will tell you if the issue is with a corrupted installation or networking issues.

Git is definitely fine on my wifi network. I use it all the time for work and hobby projects. I can tinker with the branches as you suggested but would prefer not to spend too much time on it. Are there clear instructions anywhere?

> The device only updates while offroad. Are you leaving your device on and connected to a network wifi/lte between drives?

The most recent time I tried to update, I had the car parked and on for about 20 minutes. It was "on" in the sense that the radio and electronics were on, but the engine was off.. realizing I don't know the name of this state.  

As an aside, this brings up another part of C2 UX that I've never understood: what is the C2 doing when I turn the car off and unplug it? Is it still on? I guess it might be, because it has to upload footage at some point, preferably over wifi. When I turn the car off, the screen immediately goes blank. Then I put it back in its original box, including foam. Does foam and cardboard interfere with wifi? (Clearly I'm not a hardware engineer). Anyway, some sort of indication that the C2 is still on when unplugged (instead of immediately black screen) would improve the UX IMO.

Also, I've tried plugging it into the USB-C charger for my Macbook Pro and my Chromebook in order to do the update away from the car, but it doesn't boot with either of those. Is there a recommend USB-C charger so I can just do the updates at my desk instead of sitting in the car?

VirtuallyChris, 2021-06-01:
Hey Alex, a few things.

1. The comma two has no battery so it won't be able to update or upload data while in the foam.
2. Your device should stay on even with your car off. This is what the comma power v2 is for - it provides 12v power to the device while your car is off. If that isn't working even with the proper connections, email support@comma.ai so we can get that resolved.
3. The comma two will only work inside via a USB-A to USB-C cable and a high amerage charger such as a Samsung Fast Charger.

alexklibisz, 2021-06-01:
So if I arrive at home, turn the car completely off, but don't unplug the comma, is it still on?

pd0wm, 2021-06-01:
If you see the home screen when you tap it, it's on. The device should stay on when you turn off the car. This requires the comma power being plugged into the OBD port.

alexklibisz, 2021-06-01:
Got it. Thanks, that's news to me :). I have always unplugged it immediately upon arriving home -- I park in the street and don't want to leave a fancy computer on my windshield. This probably also explains why I've had trouble finding recent drives and footage in the app. I guess I had assumed there is some minimal battery in the device that keeps it running while it's unplugged. I'll get a cable so I can boot it up inside.

adeebshihadeh, 2021-06-01:
I agree that the updater experience is super confusing, but I don't think there's an issue with the updater here. Your device just seems to rarely be online. You say that you've used the device regularly for over a year, but I only see 8 routes from your device, and only two of which are from the last month. I checked all of those routes, and none had any updater exceptions or failed updates. The updater only runs while you're offroad (car is off) - are you sure your device was connected to internet for an extended period of time while the car was off?

alexklibisz, 2021-06-01:
> The updater only runs while you're offroad (car is off) - are you sure your device was connected to internet for an extended period of time while the car was off?

The most recent time I tried to update (couple days ago), the car was off (engine off, electronics on), device was plugged in, connected to good wifi, for about 20 minutes. 

Otherwise, I've been unplugging it and putting it back int he box every time I turn the car off. I think that's likely the reason for having very few routes.. they're just not getting uploaded. Again, this is pretty confusing UX. There's nothing that tells me "turn the car off but don't unplug the device." I had always assumed it continues uploading stuff after I unplug it.

adeebshihadeh, 2021-06-01:
I'm going to close this issue unless you have any other concerns. In this case, the updater was working as intended, but never got an internet connection. The update process can and often takes more than 20 minutes, so I'm not concerned about that case. We'll have a button to do this in the next release. If you still see the updater misbehaving with this new knowledge and after the button's added, please open another issue.

closed 2021-06-01 by adeebshihadeh