commaai/openpilot #21130: Honda Nidec using wrong blinker message

VirtuallyChris, 2021-06-04:
**Describe the bug**
When doing lane changes on Honda Nidec, the stalk state is used. This causes the lane change alert to show for a split second and doesn't allow the convenience blinker to be used for lane change assist like on other makes. 

**How to reproduce or log data**
Get in a Nidec, try to change lanes.

**Device/Version information:**
_Route Coming Soon_
 - Dongle ID:
 - Route:
 - Timestamp:
 - Commit hash:
 - Car make/model: Honda Civic 2017

pd0wm, 2021-06-04:
Think this is also true for Honda Bosch and HKG

adeebshihadeh, 2021-06-04:
HKG should be fine.

ShaneSmiskol, 2021-06-05:
If anyone has a car with this issue, please post a route!

ChadBailey, 2021-06-15:
I'm unsure if this is the same bug, but my blinker is now only prompting for lane change about 30% of the time - 2018 honda bosch. The rest of the time I just have to fight the car to change lanes. This route has many instances of it happening (I always indicate, so every turn or lane change I had my blinker on)

053abdc1032b7ecf|2021-06-15--16-32-36

I will also note that you caught me right in the middle of setting up a dev environment, so it is indicating correctly that it's using my fork rather than the official release2 channel, but I promise I have not made any alterations from the release2 branch. I don't mind switching back and giving a new route if it's that important, just no promises on how soon (probably tomorrow).

adeebshihadeh, 2021-06-15:
@ChadBailey if you're looking to do dev work, this is a great first PR. Essentially, the issue here is that the signal we use to get the blinker state from the car is wrong. I believe in this case, the signal is actually the stalk position. You'll likely need to find a new signal using [Cabana](https://cabana.comma.ai), then set it [here](https://github.com/commaai/openpilot/blob/master/selfdrive/car/honda/carstate.py#L274).

ChadBailey, 2021-06-15:
Sounds good, I'll give it a shot!

ChadBailey, 2021-06-16:
After lots of searching, I could never isolate any other values that may be representing the blinker data. That said, I'm new to this and wouldn't be surprised if it were right in front of my face the entire time.

None the less, I may take another swing tomorrow if someone else hasn't solved it by then.

jyoung8607, 2021-06-16:
If you post a Cabana share link of a route you're willing for others to look at, you can get more eyes on the problem.

Cabana, "Get Share Link" button puts a magic link on the clipboard that anyone can use, paste it into this github bug.

pd0wm, 2021-06-16:
Did some looking on a Honda Accord, but couldn't find anything besides the stalk position.

For the future I wrote this script that prints out any "new" bitflips. Makes finding stuff like this a little easier. https://github.com/commaai/openpilot/pull/21294

ChadBailey, 2021-06-16:
Here is the special drive I did to debug this:

https://my.comma.ai/cabana/?route=053abdc1032b7ecf%7C2021-06-15--19-37-48&exp=1655402158&sig=od3e5tFljosaLcRGiVMis%2BhKawTRKixXc7f%2FZKw8XPc%3D&max=1&url=https%3A%2F%2Fchffrprivate-vzn.azureedge.net%2Fchffrprivate3%2Fv2%2F053abdc1032b7ecf%2Fa0ae5ed2d534a1903539f193386e289b_2021-06-15--19-37-48

I followed this routine:

Normal blinker usage ("clicked" into place and didn't hold)
Left blinker 10 blinks
Right blinker 10 blinks

Held stalk for blinker - did not allow to "click" into place
Left blinker 10 blinks
Right blinker 10 blinks

Quickly "bumped" stalk for "comfort blinker" that automatically runs for 3 blinks
Left blinker 3 blinks
Right blinker 3 blinks

You should be able to confirm by loading the 2018 Honda Accord and plotting the existing blinker data location to observe 2 long activations followed by a short activation for both blinkers

ChadBailey, 2021-06-16:
> Did some looking on a Honda Accord, but couldn't find anything besides the stalk position.
> 
> For the future I wrote this script that prints out any "new" bitflips. Makes finding stuff like this a little easier. #21294

How would one run this? Just execute it or would I need to first kill the running processes for panda+openpilot?

pd0wm, 2021-06-17:
No need to kill anything. It subscribes directly to boardd.

ChadBailey, 2021-06-23:
I've written an implementation that I'll be testing for a bit @ https://github.com/ChadBailey/openpilot/tree/feature/comfort_lane_change

you're free to test as well if you like (fair warning, i haven't tried it even once so it might be very broken) - for something like this i think i prefer to test for a while before i open any pull request to make sure it "feels right" and behaves as expected in all scenarios.

ChadBailey, 2021-06-23:
I've finished my first test. The good and bad news is it does absolutely nothing lol

pd0wm, 2021-06-23:
I was thinking something like this (untested): https://github.com/commaai/openpilot/pull/21387/files

ChadBailey, 2021-06-23:
> I was thinking something like this (untested): https://github.com/commaai/openpilot/pull/21387/files

This seems like a fantastic approach, but I wasn't able to determine that the blinker light data was available for my vehicle (2018 accord 1.5T). I didn't know the name though, so I'll take a look tonight and see if it's available.

In the event it's not, perhaps another way to make this more polished is to use the solution I was attempting but also have it only apply if the vehicle has the feature.

pd0wm, 2021-06-23:
I’m using the stalk position in that PR.

ChadBailey, 2021-06-23:
Ohhh I see now, sorry I didn't realize this had components for both light and stalk updates for multiple cars at first glance.

I like the solution, I think it's a more rigid approach that handles the concern at the vehicle layer which is where it should be handled.

I'm happy to test tonight and report back.

Note about "stealth lane changes":
I guess the only outstanding question would be from a user experience standpoint - would anyone lacking a comfort signal desire the ability to "stealth lane change" by fast-bumping the stalk to initiate a change with no indicator. more importantly, even if they desire it, would we want to support that. Im of the opinion it shouldn't be allowed for safety reasons (and imagine you feel the same) I just wanted to make that use case clear for us and any future people reading this.

ChadBailey, 2021-06-23:
Tested, left detailed notes in #21387 .

I can't recall if this was ever the behavior, but was surprised to find I felt it improved the lane change experience more than I was expecting it would. Thanks for opening the bug report @VirtuallyChris!

Regarding my prior concern
> blinker is now only prompting for lane change about 30% of the time

That was a complete fluke, I've never had that happen again. I must have unknowingly messed something up.

PR #21387 (2021-06-29): Honda: emulate convenience blinker from stalk
Closes #21130

- [x] Check convenience blinker duration (currently guessed at 3 seconds)
- [x] Check what happens after two short pressed of the stalk. Does the second press add to the convenience blinker time?
- [x] Test in car
closed 2021-06-29 by pd0wm