commaai/openpilot #21140: Kia Optima Spam ACC resume causes recurrent dig/flash up menu

sunilm1992, 2021-06-04:
**Describe the bug**

<When the car comes to a standstill behind another car the resume spam logic means that it comes up with a constant ding and flashing message - see this video I have uploaded to YouTube:

https://www.youtube.com/watch?v=RHx4VDDnnk4

The issue is my friends and family think that this is a hacky/dangerous solution because of the dinging/flashing messages and I want this to look more official, it's also annoying>

**How to reproduce or log data**

<Come to a standstill behind another car for > 3 seconds >

**Expected behavior**

<Car resumes without dinging noise - implement better resume logic which is already present in the XPS fork please!>

**Device/Version information (please complete the following information):**
 - Device: [Eon Gold]
 - Dongle ID: [c7f89112bca83879]
 - Route: [c7f89112bca83879|2021-06-04--16-31-48]
 - Timestamp: [17:42]
 - Version: [0.8.4]
 - Car make/model: [Kia Optima 2017]

**Additional context**

speedking456, 2021-06-04:
As you mentioned XPS has this, but I believe there was a reason for not including it in stock OP. I think...

sunilm1992, 2021-06-04:
I think he explained the code needed but said it could be integrated if needed

adeebshihadeh, 2021-06-12:
We spent quite a bit of time on resume for HKG, and the current implementation is very reliable. We'd certainly review a PR to improve the flickering, but this isn't considered a bug.

closed 2021-06-12 by adeebshihadeh